lagadic/visp #1048: SIFT detector crashes on macOS with OpenCV 4.5.5 release

fspindle, 2022-03-31:
#### System information (version)
- OpenCV => 4.5.5 (from brew) and master branch
- Operating System / Platform : macOS Monterey 12.2.1
- Compiler: Apple clang version 13.1.6 (clang-1316.0.21.2)

#### Description

In ViSP, `testKeypoint-5` and `testKeypoint-7` produce a crash when `SIFT::detectAndCompute()` is called

```
$ ./testKeyPoint-5 -c -d
-- Test on gray level images
INFORMATION: 
Here, we want to test feature detection on a pyramid of images even for features that are scale invariant to detect potential problem in ViSP.

Nb keypoints detected: 7640 for PyramidFAST method.
Nb keypoints detected: 6489 for FAST method.
Nb keypoints detected: 286 for PyramidMSER method.
Nb keypoints detected: 149 for MSER method.
Nb keypoints detected: 1965 for PyramidGFTT method.
Nb keypoints detected: 1000 for GFTT method.
Nb keypoints detected: 18 for PyramidSimpleBlob method.
Nb keypoints detected: 9 for SimpleBlob method.
Nb keypoints detected: 112 for PyramidSTAR method.
Nb keypoints detected: 112 for STAR method.
Nb keypoints detected: 8794 for PyramidAGAST method.
Nb keypoints detected: 7600 for AGAST method.
You should not use ORB with Pyramid feature detection!
Nb keypoints detected: 738 for PyramidORB method.
Nb keypoints detected: 476 for ORB method.
You should not use BRISK with Pyramid feature detection!
Nb keypoints detected: 3389 for PyramidBRISK method.
Nb keypoints detected: 3153 for BRISK method.
You should not use KAZE with Pyramid feature detection!
Nb keypoints detected: 595 for PyramidKAZE method.
Nb keypoints detected: 455 for KAZE method.
You should not use AKAZE with Pyramid feature detection!
Nb keypoints detected: 396 for PyramidAKAZE method.
Nb keypoints detected: 338 for AKAZE method.
Illegal instruction: 4

$ ./testKeyPoint-7 -c -d
-- Test on gray level images
Saving / loading learning files with binary descriptor are ok !
Illegal instruction: 4
```

This issue doesn't occur with OpenCV build from source from master branch on macOS.

CI shows that both test are working on Ubuntu and Windows with OpenCV 4.5.5 release.
The issue seems appearing only on macOS and should be fixed in the next OpenCV release.

fspindle, 2022-03-31:
The following sample code that uses only OpenCV allows to reproduce the issue:

```
$ more test-opencv-sift-detector.cpp
#include <iostream>
#include <vector>
#include <opencv2/core/core.hpp>
#include <opencv2/calib3d/calib3d.hpp>
#include <opencv2/features2d/features2d.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <opencv2/imgcodecs.hpp>
#include <opencv2/highgui.hpp>

int main(int argc, char *argv[])
{
  if (argc == 1)
  {
    std::cout << "Usage: " << argv[0] << " [image name]" << std::endl;
    return EXIT_SUCCESS;
  }
  cv::Mat img = cv::imread(argv[1], cv::IMREAD_GRAYSCALE);

  if (img.empty())
  {
    std::cout << "Could not read the image: " << argv[1] << std::endl;
    return EXIT_FAILURE;
  }

  std::cout << "Create mask" << std::endl;
  // generate all white mask for the usr image
  cv::Mat mask = cv::Mat(img.cols, img.rows, CV_8UC1, cv::Scalar(255.0));

  std::cout << "Mask created" << std::endl;

  std::cout << "Create SIFT detector" << std::endl;
  cv::Ptr<cv::SIFT> detector = cv::SIFT::create();

  // detect keypoints
  std::vector<cv::KeyPoint> keypoints;
  cv::Mat descriptors;

  std::cout << "Detect and compute descriptors" << std::endl;
  detector->detectAndCompute(img, mask, keypoints, descriptors);
  return EXIT_SUCCESS;
}
```

The `CMakeLists.txt` file:
```
$ more CMakeLists.txt 
project(test-opencv-sift)

cmake_minimum_required(VERSION 3.0)
find_package(openCV)

include_directories(${OpenCV_INCLUDE_DIRS})
add_executable(test-opencv-sift-detector test-opencv-sift-detector.cpp)
target_link_libraries(test-opencv-sift-detector ${OpenCV_LIBRARIES})
``` 

The output is the following:

```
$ ./test-opencv-sift-detector /tmp/opencv/opencv-master/samples/data/aero1.jpg
Create mask
Mask created
Create SIFT detector
Detect and compute descriptors
Illegal instruction: 4
```

closed 2022-04-01 by fspindle

sharmahimanshu22, 2023-04-12:
I am facing this issue with OpenCV build from source from master branch on macOS today.

commit 93bbe306 (2022-04-01): Disable SIFT keypoint detector with OpenCV 4.5.4 and 4.5.5 on macOS. Closes #1048