lagadic/visp #1052: Error when compiling tutorial-pose-from-planar-object.cpp

Alkrick, 2022-04-05:
I get the following error when compiling the code using cmake in tutorial-pose-from-planar-object.cpp

error: ‘computePlanarObjectPoseWithAtLeast3Points’ is not a member of ‘vpPose’
  413 |   const auto cMo = vpPose::computePlanarObjectPoseWithAtLeast3Points(obj_plane_in_color, model.keypoints(),

I tried setting "std=c++17" in compile options and editing the file vpPose.h in  /vision directory and commenting out the #if statement surrounding that part of the code but it didn't work

fspindle, 2022-04-05:
`vpPose::computePlanarObjectPoseWithAtLeast3Points()`was introduced last week in the master branch.
Your error means that you are linking with a version (maybe the last release) of ViSP that doesn't embed this new function.

The workaround is to uninstall old visp versions and use the one from github master branch.

Alkrick, 2022-04-06:
I downloaded the repository the same day I posted this issue and It is my first time using visp, I still made sure by reinstalling from the master branch again but the same problem occurs.

On both instances along with the previous error I got:
 ```
   error: ‘map’ in namespace ‘std’ does not name a template type
       71 |   std::map<Id, vpPoint> keypoints(const vpHomogeneousMatrix &cMo = {}) const;
```
upon checking the vpPose.h in the visp3/vision/ directory, I found that the both the <map<map>> include and the above function are surrounded by the following condition:
```` 
#if (VISP_CXX_STANDARD >= VISP_CXX_STANDARD_17) &&                                                                     \
    (!defined(_MSC_VER) || ((VISP_CXX_STANDARD >= VISP_CXX_STANDARD_17) && (_MSC_VER >= 1911)))
#include <map>
#include <optional>
#endif
````
Which seems to evaluate to false for me.

My exact installation process that produces this error is:

`git clone https://github.com/lagadic/visp.git`
`mkdir build && cd build`
`cmake -DCMAKE_CXX_FLAGS="-std=c++17" ..`
`make -j8`

I am on Ubuntu 20.04.

I tried using the latest release 3.5.0 and it compiled successfully, so I am using it for the meantime.

fspindle, 2022-04-06:
You don't have to add `"-std=c++17"` build option. Try a fresh build with:

```
$ rm -rf build
$ mkdir build
$ cd build
$ cmake ..
```

Check which c++ standard is used by default:
```
$ grep "standard" ViSP-third-party.txt 
    Use cxx standard:            17
```
and then try a fresh build
```
$ make -j8
```
It could be possible that on your platform c++17 is not supported.

If it doesn't build, try setting c++ standard to c++14 using:
```
$ cd build
$ cmake ../ -DUSE_CXX_STANDARD=14
$ make -j8
```

Let me know about these tests...

Alkrick, 2022-04-06:
I tried a fresh build and made sure that the default c++ standard is 17 as per your instructions but I still got the same error.
However, setting the standard to c++14 seemed to work and it compiled successfully.

fspindle, 2022-04-06:
What is your OS and version ?

Alkrick, 2022-04-08:
As mentioned above it's Ubuntu 20.04.04.

fspindle, 2022-04-19:
I'm unable to reproduce your build error on Ubuntu 20.04.04.

Can you get the last version:

```
$ cd visp_ws/visp
$ git pull
```
an try a build again
```
$ cd ../visp-build
$ make -j4
```
and if it fails attach the content of `visp-build/ViSP-third-party.txt`

Alkrick, 2022-04-21:
It compiled successfully earlier after changing the CXX Standard to c++14, switching it back to c++17 produces the same error even on the latest version, these are the contents of ViSP-third-party.txt : 
    ==========================================================
    General configuration information for ViSP 3.5.1
    
      Version control:               v3.5.0-70-g262a3fedb
    
      Platform:
        Timestamp:                   2022-04-21T11:13:56Z
        Host:                        Linux 5.13.0-39-generic x86_64
        CMake:                       3.16.3
        CMake generator:             Unix Makefiles
        CMake build tool:            /usr/bin/make
        Configuration:               Release
    
      C/C++:
        Built as dynamic libs?:      yes
        C++ Compiler:                /usr/bin/c++  (ver 9.4.0)
        C++ flags (Release):         -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
        C++ flags (Debug):           -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
        C Compiler:                  /usr/bin/cc
        C flags (Release):           -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
        C flags (Debug):             -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
        Linker flags (Release):      
        Linker flags (Debug):        
    
      ViSP modules:
        To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
        Disabled:                    -
        Disabled by dependency:      -
        Unavailable:                 java
    
      Python (for build):            /usr/bin/python
    
      Java:                          
        ant:                         NO
        JNI:                         NO
    
      Build options: 
        Build deprecated:            yes
        Build with moment combine:   no
    
      Mathematics: 
        Blas/Lapack:                 yes
        \- Use MKL:                  no
        \- Use OpenBLAS:             no
        \- Use Atlas:                yes
        \- Use Netlib:               no
        \- Use GSL:                  no
        \- Use Lapack (built-in):    no
        Use Eigen3:                  yes (ver 3.3.7)
        Use OpenCV:                  yes (ver 4.5.5)
    
      Simulator: 
        Ogre simulator: 
        \- Use Ogre3D:               yes (ver 1.9.0)
        \- Use OIS:                  yes (ver 1.3.0)
        Coin simulator: 
        \- Use Coin3D:               yes (ver 4.0.0)
        \- Use SoWin:                no
        \- Use SoXt:                 no
        \- Use SoQt:                 no
        \- Use Qt5:                  no
        \- Use Qt4:                  no
        \- Use Qt3:                  no
    
      Media I/O: 
        Use JPEG:                    yes (ver 80)
        Use PNG:                     yes (ver 1.6.37)
        \- Use ZLIB:                 yes (ver 1.2.11)
        Use OpenCV:                  yes (ver 4.5.5)
[... 62 lines cut ...]
    
      Optimization: 
        Use OpenMP:                  yes
        Use pthread:                 yes
        Use pthread (built-in):      no
        Use cxx standard:            17
    
      DNN: 
        Use CUDA Toolkit:            no
        Use TensorRT:                no
    
      Documentation: 
        Use doxygen:                 yes
        \- Use mathjax:              no
    
      Tests and samples:
        Use catch2 (built-in):       yes (ver 2.13.7)
        Tests:                       yes
        Demos:                       yes
        Examples:                    yes
        Tutorials:                   yes
    
      Install path:                  /usr/local
    
    ==========================================================

fspindle, 2022-04-21:
Thanks.

Can you also attach the full build error

Alkrick, 2022-05-10:
I apologize for the late response, I was unable to access my workstation for a while.

The full build is attached [here](https://github.com/lagadic/visp/files/8657522/buildError.txt) and the output of the command you provided is as follows:

    #define VISP_CXX_STANDARD_98 199711L
    #define VISP_CXX_STANDARD_11 201103L
    #define VISP_CXX_STANDARD_14 201402L
    #define VISP_CXX_STANDARD_17 201703L
    #define VISP_CXX_STANDARD 201703L
    #if (VISP_CXX_STANDARD >= VISP_CXX_STANDARD_11)

fspindle, 2022-05-11:
@Alkrick Incredible!

Can you add `#include <map>` in `tutorial/computer-vision/tutorial-pose-from-planar-object.cpp` at line 2 and try a new build.

fspindle, 2022-05-18:
@Alkrick I'm quite sure that you have an older visp version installed in `/usr` or `/opt/ros` and that the tutorial is going to be linked with this version of visp that doesn't embed `vpPose::computePlanarObjectPoseWithAtLeast3Points()`.

1. First you need to identify if you have multiple visp versions installed

    ```
    $ sudo updatedb
    $ locate VISPConfig.cmake
    ```
    At this point, if I'm right you should see a `VISPConfig.cmake` file located in `/opt/ros` or `/usr`

2. Uninstall visp from `/opt/ros` or `/usr`
    Here you can use `synaptic` to remove `ros-noetic-visp` or `libvisp-dev` packages 

3. Clean potential changes introduced in `visp_ws/visp`
4. Try a new build 
    ```
    $ cd visp_ws/visp-build
    $ make -j4
    ```

fspindle, 2022-06-20:
@Alkrick Any update ? Otherwise will be closed

Alkrick, 2022-06-27:
@fspindle Unfortunately I won't be able to access the workstation on which this error occurred for a while. You are right I have a different version installed but I doubt that its the cause of the error, since I installed it after getting the error (as mentioned by the last line on my second entry to this thread)

fspindle, 2022-06-27:
@Alkrick Ok I will close the thread for now. Don't hesitate to open a new one if you are able to reproduce the error. I'm 100% sure that it comes from multiple ViSP versions installed on your system.

closed 2022-06-27 by fspindle

lijie942, 2023-03-21:
First of all, thank you very much for your help, I did make the mistake you pointed out - installing multiple versions of visp    ros-noetic-visp.