lagadic/visp #1054: Some question about photometric moments visual serving

maoni99999, 2022-04-18:
Thank you very much for the open source code of visual servoing provided by you. I am currently conducting research on visual servoing using image moment. By reading the source code related to image moment servoing, I feel the complexity of moment feature and its corresponding interaction matrix analytic form calculation At the same time, I also feel the advantages of image moment servo, such as large convergence range, strong decoupling ability, etc., but I recently found an article called ''A Direct Dense Visual Servoing Approach using Photometric Moments'' https://ieeexplore.ieee.org/abstract/document/8372943, which is from Manikandan Bakthavatchalam . The article gives the analytical form of the visual servo of the photometric moment servo with spatial weighting function. I read the article carefully. If I want to change it to the form of this article, it involves changes to the basic moment m00, m01, etc., high-order moment features and All the corresponding interaction matrices have to be changed. Because of the heavy workload, I would like to ask if you have added relevant code changes to be released. If so, I would like to thank you very much if you can provide the corresponding code. If not I will take the time to complete the code implementation of the photometric moment servo with weights and share it with you, thanks.

s-trinh, 2022-04-20:
@fspindle 

Github offers the possibility to use Discussion tab:
- https://github.com/zeux/pugixml/discussions
- https://github.com/apache/incubator-mxnet/discussions

Maybe this kind of question can be move to discussion.

chaumett, 2022-04-20:
You are fully right. Up to now, only the non-weighted photometric moments are available in ViSP. We do not plan to release the code for the weighted moments due to their complexity. We are also not fully satisfied with their results since they do not preserve invariance properties. Good luck to improve that work!