lagadic/visp #1070: Extend FOV on realsense D435i

parkoh25, 2022-06-17:
Extend the stream of realsense D435i

I'm curious about the code that extends the stream of realsense D435i.

The example provided is 640X480, and changing the code does not change and does not run.

I'd like to check the operation on a wider stream.

jnader, 2022-06-17:
Hello @parkoh25 ,

Concerning the wider stream of Intel RealSense D435, you can check [testRealSense2_D435.cpp](https://visp-doc.inria.fr/doxygen/visp-daily/testRealSense2_D435_8cpp-example.html). In this test example, you activate the `RS2_STREAM_COLOR` with `1280x720` resolution (the same goes for depth and infrared streams).

You can check the source code in `$VISP_WS/modules/sensor/test/testRealSense2_D435.cpp`, and if you want to run it you will find it in your `build` directory in `modules/sensor`.

Note that this example is using `D435` and not `D435i` (which has an internal IMU).

Concerning the AprilTag, the output is a `vpHomogeneousMatrix` including translation and rotation between camera's and tag's frames.
Here is an [example](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-apriltag-detector-live-rgbd-realsense_8cpp-example.html) for detecting apriltag with realsense D435. You can check how to display information on the image using `vpDisplay::displayFrame()` and `vpDisplay::displayText()`.

parkoh25, 2022-06-18:
Thank you for your detailed explanation.

Can I ask you one more question?

I will make the object's detection with point cloud data into a homogeneous matrix. I would appreciate it if you could give me an answer to this.

jnader, 2022-06-20:
I am not sure I understand exactly your question.
What type of object are you trying to detect with point cloud?

jnader, 2022-07-06:
@parkoh25 

To the best of my knowledge, in order to perform PBVS you should have an estimate of the pose of the object relative to the camera.

parkoh25, 2022-07-08:
@jnader

Then, is PBVS possible only with the location data of the object?  I understand that rotation should also be considered.
If possible, which part of the code should I enter the position data?

jnader, 2022-07-08:
You need the pose (position + rotation)

closed 2022-07-21 by fspindle