lagadic/visp #1085: Inquiries about using Real Sense Camera

parkoh25, 2022-08-02:
Hello

I am using the real sense camera D435i.

So I am using vprealsense.h provided by visp.

However, I understand that it should not be used at the same time as the librealsense provided by Real Sense.

I want to use both the vprealsense::open() code and rs2::pipeline at once. vprealsense::open() and rs2::pipeline.start() were not always at the same time. Could you tell me how to use it together? I understand that both codes use librealsense.

s-trinh, 2022-08-04:
Have a look at the different tests (https://github.com/lagadic/visp/tree/master/modules/sensor/test/rgb-depth) or tutorial code (https://github.com/lagadic/visp/tree/master/tutorial/grabber) to see how to use librealsense and ViSP.

s-trinh, 2022-08-04:
See:
- [`getPipeline()`](https://visp-doc.inria.fr/doxygen/visp-daily/classvpRealSense2.html#acbf6bbeea5cc26a7e78a305b5abafe4c)
- [`getPipelineProfile()`](https://visp-doc.inria.fr/doxygen/visp-daily/classvpRealSense2.html#a47605953f3ba77db27f41c3427bbd992)

You should be able to use simultaneously the ViSP or the librealsense "backend".