lagadic/visp #1089: Symbol lookup and undefined synbol error

mkaur013, 2022-09-02:
I am getting below errors after install visp and visp-ros:- 
"./servoUniversalRobotsIBVS: symbol lookup error: ./servoUniversalRobotsIBVS: undefined symbol: _ZN12vpRealSense2C1Ev"

"./testUniversalRobotsGetData: symbol lookup error: ./testUniversalRobotsGetData: undefined symbol: _ZN22vpRobotUniversalRobotsC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE"

I tired to look at the issue following " Symbol Lookup Error: using visp generic-rgbd-realsense module #766 "
After running ```$ sudo updatedb```
I found version issue 
```/visp-ws/visp-build/lib/libvisp_core.so.3.5.1 and /opt/ros/noetic/lib/libvisp_core.3.5.0```
After uninstalling visp and visp-ros
Downloaded visp.3.5.0 and visp-ros

Still same issue
I would really appreciate any assistance with this problem. Thank you!

s-trinh, 2022-09-05:
You should keep only one version of ViSP. If you need ViSP built with some specific libraries, you have to remove ViSP installed by ROS and just keep yours.
For ROS, you can then clone the repositories you need and build them with the correct ViSP version in your `catkin` workspace.

mkaur013, 2022-09-06:
ViSP version 3.5.0 works fine. I have tested controlling robot using april tag and it works. Thanks for the tutorial and all the guide. 

In my VISP_WS, I have clone visp3.5.0 and build it. Then created new folder Vision_Visp/src. Clone and build vision_visp and visp_ros in Vision_visp ```catkin``` workspace and the lib version is ```libvisp_core.so.3.5.0```

After building the workspace, I start getting errors.

rolalaro, 2024-06-03:
Hi,

Have you tried to check in visp_sensor library if there are undefined symbols ?

```
$ cd visp_ws/visp-build/lib
$ nm -gDC libvisp_sensor.so.3.6.1 | grep "U vpRealSense2"
```

closed 2025-11-03 by fspindle