lagadic/visp #1100: Cannot comunicate with UR5 robot using ur_rtde library

aaflakiyan, 2022-09-29:
Hi everyone, 
When I try to test getting data from my UR5 robot I run into this issue. Could you please help me solve that?

$ ./testUniversalRobotsGetData --ip <my_robot_ip>
-- Start test 1/3
ur_rtde exception: The following variables was not found by the controller: [payload_inertia],
 see which variables are supported by your PolyScope version here: 
https://www.universal-robots.com/articles/ur/interface-communication/real-time-data-exchange-rtde-guide

wine3603, 2022-12-08:
hi @fspindle , I met a similar problem:
I followed the tutorial, built the ur_rtde with visp and get this:

> a@xx:~/visp-ws/visp-build$ grep ur_rtde ViSP-third-party.txt
    Use ur_rtde:                 yes (ver 1.5.5)

$ ./testUniversalRobotsGetData --ip <my_robot_ip>

fspindle, 2022-12-08:
This is a classical error. You have a version of ViSP that doesn't support `liburtde` installed in `/usr` (typically the one coming with `libvisp-dev` or `ros-distro-visp` debian package).

To check
```
$ sudo updatedb
$ locate libvisp_robot.so
```
The solution is to remove the one installed in `/usr`, or if this is not possible to build from source with support of additional 3rd parties (like in your case `libur_rtde`) and then install that version in the same location as the one in `/usr`.

wine3603, 2022-12-09:
Hi Dr. Spindle, thank you so much for your quick response!
```
$ locate libvisp_robot.so
/home/a/visp-ws/visp-build/lib/libvisp_robot.so
/home/a/visp-ws/visp-build/lib/libvisp_robot.so.3.5
/home/a/visp-ws/visp-build/lib/libvisp_robot.so.3.5.1
/opt/ros/melodic/lib/libvisp_robot.so
/opt/ros/melodic/lib/libvisp_robot.so.3.5
/opt/ros/melodic/lib/libvisp_robot.so.3.5.0
/usr/local/lib/libvisp_robot.so
/usr/local/lib/libvisp_robot.so.3.5
/usr/local/lib/libvisp_robot.so.3.5.1

```

fspindle, 2022-12-09:
After installing `ur_rtde` in `/usr/local` you should build ViSP again.

wine3603, 2022-12-09:
Dear Prof. @fspindle Thank you again, you solved my problem. 
I did cmake ../visp, forget to make it again :point_up_2:

closed 2022-12-09 by fspindle

wine3603, 2022-12-09:
Hi @fspindle , I got a new problem :
```
$ ./modules/robot/testUniversalRobotsGetData 
-- Start test 1/3
Robot connected  : yes
Robot mode       : 3
Robot model      : could not understand: 'get robot model'
PolyScope version: 3.10.0.76181
-- Start test 2/3
RTDEControlInterface: Could not receive data from robot...
RTDEControlInterface Exception: Operation canceled
Reconnecting...
RTDEControlInterface Exception: write: Bad file descriptor
terminate called after throwing an instance of 'std::runtime_error'
  what():  RTDEControlInterface: Could not reconnect to robot...
```
I tried to update UR5's polyscope, but there was a problem of the robot  unpacking the ur5 update file(xxx.urup), any help?