lagadic/visp #1109: problem about :change the control mode from stop to velocity control

867415148, 2022-10-14:
I follow the tutorial：IBVS with a robot from Universal robot，try to complete  visual servo.  
When I run this command： ./servoUniversalRobotsIBVS --eMc ../../tutorial/calibration/ur_eMc.yaml --plot。 My UR5  moves to a certain position，then the  system prompts me：change the control mode from stop to velocity control。After that, my UR5 has no reaction.
Before I run this command，everthing is ok。
The system prompt is as shown in the figure。
The pose of UR5 after moving to the certain position is shown in the figure.
How can I solve this problem？ Help!!!!!!!!
[image]
[image]

fspindle, 2022-10-14:
The msg `Change the control mode from stop to velocity control` is just here to warn the user that the robot will be controlled in velocity.

The `servoUniversalRobotsIBVS` binary should open a windows that is displaying the image acquired by the Realsense camera. The robot can then only move if the apriltag is detected in the image and if the user enables the controller by a left click in the image.

867415148, 2022-10-15:
Thank you so much. I completed this experiment, and the effect is very good.Thank you again for your contribution and guidance.

closed 2022-10-28 by fspindle