lagadic/visp #1114: OpenCV, Eigen and FlyCapture libraries not detected by VIPS on Windows10

Ivan-Jordane, 2022-11-17:
I have installed VISP from source for Windows 10 with Visual C++ 2019 (vc16). And I have installed also OpenCV, Eigen, and FlyCapture as 3rd parties, following all the steps of the website.
1.	I can see OpenCV and Eigen as 3dr parties in the ViSP-third-party.txt file, but not FlyCapture,
2.	When I tried to run some Examples from VISP, OpenCV and Eigen are not detected.
Thanks for your help!
Best regards.

[image]
[image]

[image]

fspindle, 2022-11-18:
From what I see, you have a build error with Eigen. I suspect that you don't follow carefully all the steps to install Eigen (see [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc17.html#install_win10_msvc17_3rdparty_eigen)), especially the command to install Eigen
```
cmake --build . --config Release --target install
```

Concerning Flycapture, did you carefully follow the [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc17.html#install_win10_msvc17_3rdparty_flycapture) and complete Flycapture SDK installation by adding `C:\Program Files\Point Grey Research\FlyCapture2\bin64` to the `Path` environment var ?

Don't forget that after modifying the `Path`env var, you need to quit `cmake-gui` and all `cmd` windows, before opening them again so that the env vars are taken into account.

Ivan-Jordan96, 2022-11-18:
Thank you for your quick response.
I will try to reinstall VISP and the different third parties.
Please, is it mandatory to install Flycapture SDK in  C:\Program Files?
Best Regards.

fspindle, 2022-11-18:
You can install Flycapture SDK where you want, but you have then:
- to add the location of the DLL in the `Path` var
- add a new var named `FLYCAPTURE_HOME` that points to your other installation location (see [here](https://github.com/lagadic/visp/blob/master/cmake/FindFlyCapture.cmake#L49))
- I suspect that you will also have to modiy [this line](https://github.com/lagadic/visp/blob/master/cmake/FindFlyCapture.cmake#L52) and [this line](https://github.com/lagadic/visp/blob/master/cmake/FindFlyCapture.cmake#L57), by using rather `list(APPEND FLYCAPTURE_INC_SEARCH_PATH "$ENV{FLYCAPTURE_HOME}/include")`

Let me know, so that I can commit the changes in the code.

Ivan-Jordan96, 2022-11-20:
Thank you again for these methods, everything works now even FlyCapture. But VIPS doesn't detect my camera and I will try to solve that. In this image, you can see what I got when I run _tutorial-grabber-flycapture_. 
Best regards.

[image]

Ivan-Jordan96, 2022-11-20:
I want to specify that I used a BlackFly color camera S UBS 3.1 BFS-U3-63S4C-C.