lagadic/visp #1115: Install Spinnaker® SDK as a third party of VISP

Ivan-Jordan96, 2022-11-23:
Dear Madam/Sir,

I tried to use a BlackFly color camera (S UBS 3.1 BFS-U3-63S4C-C) with VISP, but FlyCapture didn't detect my camera. But the software Spinnaker® SDK works very well with the camera, so I want to know if it is possible to install Spinnaker® SDK as a third party of VISP. 
In the case it is possible, please how to do it.
Thank you, best regards.

fspindle, 2022-11-23:
ViSP doesn't support Spinnaker SDK yet and since I don't have a FLIR camera under the hand, I cannot introduce this 3rd party.

If your camera can be used with FlyCapture, it should be possible to use it with ViSP, otherwise you can contribute to ViSP by getting inspiration from [vpFlyCaptureGrabber](https://github.com/lagadic/visp/blob/master/modules/sensor/src/framegrabber/flycapture/vpFlyCaptureGrabber.cpp) and implementing something equivalent in a new vpSpinnakerGrabber class.

Ivan-Jordan96, 2022-12-07:
Dear Sir,
After the discussion we had above, I tried to create a new vpSpinnakerGrabber class. I built everything without a problem, but when I tried to run a program based on this class I got an error message, as you can see in the picture below.  
Even for the very simple class that you can also see below, I got the same message.
Please, I need some help or explanation to solve that.
Thank you, best regards.

[image]
[image]
[image]
[image]

fspindle, 2022-12-08:
Everything sounds ok in the code.
I suspect that you are using `libvisp_sensor` dll that is not build with `vpSpinnakerGrabber` class. This could be due to the `Path` env var that points for example to an install folder like `%VISP_WS%/visp-build-vc17/install/x64/vc17/bin`.

If this is the case, build `INSTALL` target to install your modified version in  `%VISP_WS%/visp-build-vc17/install/x64/vc17/bin`. Since `INSTALL` is quite long, instead you can copy `testSpinnakerSDK.exe` in  `visp-build-vc17/x64/vc17/bin/Release` (the location of the built dll) and run the test to see if it works.

closed 2022-12-08 by fspindle