lagadic/visp #1125: Issue with tracking tutorials

Kaleye, 2023-02-02:
Hello when I try to run any of the CAD model based trackers based on the tutorials, I get this error:
erika@erika-linux:~/visp-ws/visp-build/tutorial/tracking/model-based/generic$ sudo ./tutorial-mb-generic-tracker-full --video $VISP_WS/cubesat/video/00%2d.png --model $VISP_WS/cubesat/cubesat1b.cao
[sudo] password for erika: 
Video name: /home/erika/visp-ws/cubesat/video/00%2d.png
Tracker requested config files: /home/erika/visp-ws/cubesat/cubesat1b.[init,xml,cao or wrl]
Tracker optional config files: /home/erika/visp-ws/cubesat/cubesat1b.[ppm]
 *********** Parsing XML for ME projection error ************ 
projection_error : sample_step : 10 (default)
projection_error : kernel_size : 5x5 (default)
 *********** Parsing XML for Edge Model-Based Tracker ************ 
ecm : mask : size : 5
ecm : mask : nb_mask : 180
ecm : range : tracking : 5
ecm : contrast : threshold 500
ecm : contrast : mu1 0.5
ecm : contrast : mu2 0.5
ecm : sample : sample_step : 4
camera : u0 : 320
camera : v0 : 240
camera : px : 1199.99
camera : py : 1199.99
face : Angle Appear : 80
face : Angle Disappear : 85
face : Near Clipping : 0.01
face : Far Clipping : 100
face : Fov Clipping : True
lod : use lod : 0 (default)
lod : min line length threshold : 50 (default)
lod : min polygon area threshold : 2500 (default)
> 188 points
> 0 lines
> 0 polygon lines
> 13 polygon points
> 0 cylinders
> 0 circles
Cannot read /home/erika/visp-ws/cubesat/cubesat1b.0.pos
cMo set to identity
Load 3D points from: /home/erika/visp-ws/cubesat/cubesat1b.0.pos/home/erika/visp-ws/cubesat/cubesat1b.init
Cannot read /home/erika/visp-ws/cubesat/cubesat1b.0.pos/home/erika/visp-ws/cubesat/cubesat1b.init
Catch a ViSP exception: Error [3]:	Cannot open model-based tracker init file /home/erika/visp-ws/cubesat/cubesat1b.0.pos/home/erika/visp-ws/cubesat/cubesat1b.init

Any ideas what's wrong?

fspindle, 2023-02-03:
I'm unable to reproduce the error!
Can you try the following again:

```
$ cd $VISP_WS/visp-build/tutorial/tracking/model-based/generic
$ wget http://visp-doc.inria.fr/download/mbt-model/cubesat.zip
$ unzip cubesat.zip
$ ./tutorial-mb-generic-tracker-full --video cubesat/video/00%2d.png --model cubesat/cubesat1b.cao
```
It should open a windows with the first b&w image of the video and an other window showing a color image with the location of 4 red crosses numbered from 1 to 4.
- In the b&w image you have to left click on point 1, 2, 3 and 4.
- Now the cad model should appear in green in overlay
- Then left click to start the tracking

Kaleye, 2023-02-03:
Okay I tried that and it briefly opens the image, but then crashes and shows basically the same error:

Video name: cubesat/video/00%2d.png
Tracker requested config files: cubesat/cubesat1b.[init,xml,cao or wrl]
Tracker optional config files: cubesat/cubesat1b.[ppm]
 *********** Parsing XML for ME projection error ************ 
projection_error : sample_step : 10 (default)
projection_error : kernel_size : 5x5 (default)
 *********** Parsing XML for Edge Model-Based Tracker ************ 
ecm : mask : size : 5
ecm : mask : nb_mask : 180
ecm : range : tracking : 5
ecm : contrast : threshold 500
ecm : contrast : mu1 0.5
ecm : contrast : mu2 0.5
ecm : sample : sample_step : 4
camera : u0 : 320
camera : v0 : 240
camera : px : 1199.99
camera : py : 1199.99
face : Angle Appear : 80
face : Angle Disappear : 85
face : Near Clipping : 0.01
face : Far Clipping : 100
face : Fov Clipping : True
lod : use lod : 0 (default)
lod : min line length threshold : 50 (default)
lod : min polygon area threshold : 2500 (default)
> 188 points
> 0 lines
> 0 polygon lines
> 13 polygon points
> 0 cylinders
> 0 circles
Cannot read cubesat/cubesat1b.0.pos
cMo set to identity
Load 3D points from: cubesat/cubesat1b.0.poscubesat/cubesat1b.init
Cannot read cubesat/cubesat1b.0.poscubesat/cubesat1b.init
Catch a ViSP exception: Error [3]:	Cannot open model-based tracker init file cubesat/cubesat1b.0.poscubesat/cubesat1b.init

fspindle, 2023-02-03:
Which version of ViSP do you have ?
If this is not the case, try cloning ViSP from github https://github.com/lagadic/visp

Kaleye, 2023-02-03:
I cloned visp from github on Ubuntu LTS 20.04 then used the tutorials to get it working.

closed 2023-02-23 by fspindle