lagadic/visp #1127: Dll not found after reinstallation of VISP - visp_io351.dll not found

VarnL0, 2023-02-16:
I have an error right after the installation of Visp and dataset, when I try to run the first example with displayGDI.exe I end up with this error :

visp_io351.dll not found. Reinstalling the program can help to solve this problem.

I reinstalled the program and the dataset but I end up with the exact same error at the same moment even though the installation worked perfectly.

Also the dll file is existing, could it help to move it to another directory ?
[image]
[image]

fspindle, 2023-02-16:
You forgot to add the location of ViSP DLL in your path env var as explained in the [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc17.html#install_win10_msvc17_env_var)

In your case you should add `C:\visp-ws-premier-essai\visp-build-vc17\install\x64\vc17\bin`

closed 2023-02-16 by fspindle

VarnL0, 2023-02-16:
I actually added the location as mentionned in the tutorial, the problem is maybe coming from something else...
[image]

fspindle, 2023-02-16:
You have to open a fresh cmd Windows after changing the env var.

VarnL0, 2023-02-17:
Oh yes thank you  now everything works fine !