lagadic/visp #1184: Getting "ur_rtde exception: Couldn't resolve requests" when executing servoUniversalRobotsIBVS

DTSKT, 2023-06-29:
Hi,

I'm following the tutorial step by step and after the calibration of both intrinsic and extinsic parameters, I tried the IBVS executing: ./servoUniversalRobotsIBVS --eMc ../../tutorial/calibration/ur_eMc.yaml --plot, but It throws thr exception: ur_rtde exception: Couldn't resolve requests.

I'm using native OS Ubuntu 20.04.5 LTS, and I've installed all the necessary librarys and packages, as I said, I was able to perform all the calibration, using a camera Intel Realsene L515.

After executing the binary It doesn't even open a new window showing the image of the camera as said in the tutorial, so I don't really now where can the error come from.

Thanks.

DTSKT, 2023-07-03:
After further experimentation I found that the error must be related with the camera, in the part of the code showed. If I disable the depth and infrared stream and only enable color, the error doesn't occur, so It must be related with depth and infrarred. As I said, the camera I'm using is a realsense L515. I know the hardware is in good shape as the example of librealsense that shows all the streams works properly, showing accel, gyro, depth and infrarred on screen.

vpRealSense2 rs;
rs2::config config;
unsigned int width = 640, height = 480;
config.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGBA8, 30);
config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
config.enable_stream(RS2_STREAM_INFRARED, 640, 240, RS2_FORMAT_Y8, 30);

rs.open(config);

fspindle, 2023-07-13:
Sorry, I don't have an L515 camera to test. Try to disable infrared...

fspindle, 2025-11-03:
I will close this issue. If needed reopen it

closed 2025-11-03 by fspindle