lagadic/visp #1191: Megapose tutorial IndexError: list index out of range error

mattratcliffe, 2023-07-08:
Hi there

I am working through the Tracking with MegaPose tutorial and I hit an error when trying to execute the install.py script. 

I am running Windows 11 Build - 22621.1848.

https://github.com/lagadic/visp/blob/master/script/megapose_server/install.py

```
(base) C:\visp-ws\visp\script\megapose_server>python install.py
C:\visp-ws\visp\script\megapose_server\install.py:13: SyntaxWarning: assertion is always true, perhaps remove parentheses?
  assert(len(megapose_env_line) == 1, 'Found multiple environment names with same name, shouldnt happen')

This script installs Megapose and the server to communicate with ViSP.
the file "megapose_variables.json" specifies where megapose should be cloned and where the models should be downloaded.
It also contains the name of the conda environment to create.

current values:
  - Megapose directory: C:\visp-ws\visp\script\megapose_server\megapose6d
  - Megapose model directory: C:\visp-ws\visp\script\megapose_server\megapose6d\data
  - Conda environment name: visp_megapose_server

Installation requires:
  - git
  - conda

All these programs should be in your path.

It will create a new environment called megapose, with all dependencies installed.
All the megapose models will be downloaded via rclone.

If you encounter any problem, see https://github.com/megapose6d/megapose6d for the installation steps.

The steps followed in the script are the same, but end with the installation of the megapose_server package (the python scripts in this folder)

Cloning megapose directory...
Cloning megapose git repo...
Cloning into 'C:\visp-ws\visp\script\megapose_server\megapose6d'...
remote: Enumerating objects: 202, done.
remote: Counting objects: 100% (202/202), done.
remote: Compressing objects: 100% (166/166), done.
remote: Total 202 (delta 42), reused 181 (delta 28), pack-reused 0Receiving objects:  79% (160/202), 10.55 MiB | 20.64 MReceiving objects:  80% (162/202), 10.55 MiB | 20.64 MiB/s
Receiving objects: 100% (202/202), 19.54 MiB | 21.58 MiB/s, done.
Resolving deltas: 100% (42/42), done.
Submodule 'deps/bop_toolkit_challenge' (https://github.com/ylabbe/bop_toolkit_challenge) registered for path 'deps/bop_toolkit_challenge'
Cloning into 'C:/visp-ws/visp/script/megapose_server/megapose6d/deps/bop_toolkit_challenge'...
Submodule path 'deps/bop_toolkit_challenge': checked out '55f3bf1f70d7a85e846e747cde19633117bfe0df'
Creating conda environment and installing megapose...
Traceback (most recent call last):
  File "C:\visp-ws\visp\script\megapose_server\install.py", line 117, in <module>
    install_dependencies(megapose_dir, megapose_environment)
  File "C:\visp-ws\visp\script\megapose_server\install.py", line 48, in install_dependencies
    if not conda_env_already_exists(megapose_environment):
  File "C:\visp-ws\visp\script\megapose_server\install.py", line 29, in conda_env_already_exists
    return get_megapose_bin_conda_env(megapose_env).exists()
  File "C:\visp-ws\visp\script\megapose_server\install.py", line 14, in get_megapose_bin_conda_env
    megapose_env_line = megapose_env_line[0]
IndexError: list index out of range
```
I have installed Conda and have the correct version.

I'm quite new to Python so unsure how to proceed. Any help would be much appreciated. 

My megapose_variables.json file has the following:
```
 "megapose_dir": "./megapose6d",
  "megapose_data_dir": "./megapose6d/data",
  "environment": "visp_megapose_server"
```

thanks
Matt

SamFlt, 2023-07-12:
Hi Matt,

Thank you for raising an issue!

As a first step for debugging, could you include the output of the command `conda info --envs`?

It should look something like:
```
# conda environments:
#
base                     /local/sfelton/miniconda3
blenderproc              /local/sfelton/miniconda3/envs/blenderproc
cosypose                 /local/sfelton/miniconda3/envs/cosypose
megapose_visp            /local/sfelton/miniconda3/envs/megapose_visp
nerfstudio            *  /local/sfelton/miniconda3/envs/nerfstudio
yolov7                   /local/sfelton/miniconda3/envs/yolov7
```

If the command does not work, (conda is not found), you may need to add it to your PATH variable and restart your terminal. It seems that if you installed conda for all users, this step is required (see [Here](https://docs.conda.io/projects/conda/en/latest/user-guide/install/windows.html))
If this was not done before, you should also run  `conda init powershell` to setup your environment for conda.

Cheers,
Sam

fspindle, 2023-07-13:
@mattratcliffe Can you test the PR: https://github.com/lagadic/visp/pull/1197 and give us a feedback.

mattratcliffe, 2023-07-13:
Thanks @SamFlt @fspindle for the responses. I've tested the PR: https://github.com/lagadic/visp/pull/1197 and can confirm it fixes the issue. I can successfully activate the megapose environment. 

```
Traceback (most recent call last):
  File "C:\visp-ws\visp\script\megapose_server\install.py", line 124, in <module>
    install_dependencies(megapose_dir, megapose_environment)
  File "C:\visp-ws\visp\script\megapose_server\install.py", line 63, in install_dependencies
    subprocess.run([megapose_env_pip, 'install', '-e',  str(megapose_path)], check=True)
  File "C:\Users\user_1\anaconda3\lib\subprocess.py", line 503, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\user_1\anaconda3\lib\subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\user_1\anaconda3\lib\subprocess.py", line 1440, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

(base) C:\visp-ws\visp\script\megapose_server>conda activate visp_megapose_server

(visp_megapose_server) C:\visp-ws\visp\script\megapose_server>conda env list
# conda environments:
#
base                     C:\Users\user_1\anaconda3
visp_megapose_server  *  C:\Users\user_1\anaconda3\envs\visp_megapose_server
```

thanks
Matt

closed 2023-07-13 by mattratcliffe