lagadic/visp #1203: Access violation error:tutorial-apriltag-detector-live

Hawthorne0, 2023-07-15:
Hello everyone. When I follow the tutorial of AprilTag marker detection(https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-apriltag.html),the program has encountered access violation error. How can I fix it.
[image]

yangqingnian, 2023-07-17:
i can't solve the problem

Hawthorne0, 2023-07-17:
> Which OS ? Which Compiler ? Which ViSP version ? Which grabber used in tutorial-apriltag-detector-live.cpp ? Does tutorial-apriltag-detector.cpp work ?

windows 11,visual studio 2022,visp 3.5.0, OpenCV and V4l2 grabber,Both programs dont work with the same error .Any help would be greatly appreciated.

Hawthorne0, 2023-07-17:
> The first thing to do is to try with visp github version. Check also that the location of ViSP dll files are in the Path env var.
That's right. When I encountered this error, I upgraded visp using github version firstly. But it still doesnot work with the same error. The location of ViSP dll files was also checked.  
I guess maybe this line: detector.detect(I, tagSize, cam, cMo_vec) of  tutorial-apriltag-detector-live.cpp arised unexpected problems cause the program could work without error by deleteing this line. I dont know how to fix it.

fspindle, 2023-07-17:
There is something wrong in your setup. What is the content of your `ViSP-third-party.txt file` ?

Try to do a fresh visp build without 3rd parties and run tutorial-apriltag-detector.cpp again.
If it doesnt' work use the debugger to figure out the issue.

Hawthorne0, 2023-07-17:
Okay, thanks a lot.

yangqingnian, 2023-07-17:
good @Hawthorne0

closed 2025-11-03 by fspindle