lagadic/visp #1218: vpMbtDistanceLine remove from tracking

constiDisch, 2023-08-25:
Hi,
i am quite new to visp and have the following question:
I have a relatively complex cad model, which I wanted to load and track with the `vpMbGenericTracker` in particular the moving edge approach.
However, I want more or less all faces in a visibility check, but be able to remove single lines individually based on other criteria (e.g. to use only lines which have faces with a large angle, or to focus on some very specific part if the model if visible, ...).

So my idea was to loop through the `vpMbtDistanceLine` and remove them from tracking (without removing the entire face from tracking, because this could have some side-effects to other lines).

Is there a way to achieve this? 
I found `vpMbtDistanceLine.setTracked` --> but it has some checks with the connected polygons. 
Thanks a lot for you hints.

fspindle, 2023-08-28:
I suggest that you have a look to [How to set a name to a face](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic.html#mb_generic_advanced_cao_nam) and [How not to consider specific polygons](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic.html#mb_generic_advanced_exclude).

constiDisch, 2023-08-29:
Thanks for your hints. Perhaps I am missing something here, but I do not want to set an entire polygon  as not tracked, but only a single line of polygon.
Thus I was looking at the code in `vpMbtDistanceLine::setTracked`, but it seems to check whether all corresponding polygons are not tracked and then turn off the tracking. 
So is there a possibility to set tracked/not for an single edge? 
I think of something like an forceTrack (or similar)?

constiDisch, 2023-09-05:
Thanks for your help and suggestions. I will have a try!