lagadic/visp #1288: error connecting with UR10e robotundefined symbol: _ZNK22vpRobotUniversalRobots12getRobotModeEv

shrutichakraborty, 2023-12-05:
Hello! I have followed the tutorial on https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-calibration-extrinsic.html

But when I launch the file `./testUniversalRobotsGetData --ip <my robot ip >` I get the following error:

./testUniversalRobotsGetData: symbol lookup error: ./testUniversalRobotsGetData: undefined symbol: _ZNK22vpRobotUniversalRobots12getRobotModeEv 

I am running the files on an ubuntu22.04 machine and I have installed and built visp from source. Any ideas would be very helpful?

fspindle, 2023-12-05:
You can launch `ldd ./testUniversalRobotsGetData` and see which is the `libvisp_robot.so` library that is used.

One possible reason is that you have multiple ViSP libraries installed and the one that is used isn't build with `ur_rtde` 3rd party. To check that, run:

```
$ sudo updatedb 
$ locate libvisp_robot.so
```

PS: Run `sudo apt install locate` if `updatedb` is unknown.

shrutichakraborty, 2023-12-05:
Hello @fspindle, thanks! I have visp-ros installed as well.. I'm not sure if that is the issue. Here is what I find when I execute `locate libvisp_robot.so`

[image]

Here's what I have when I run `ldd ./testUniversalRobotsGetData`
[image]

I guess I made a listake with my setup, however, I had followed the instructions to build my visp_ws from source (as it said to do so to work with real robot) and prior to that I had installed visp-ros and visp_vision but these folders are in my ros2 workspace. 

Could you tell me how to solve this issue?

fspindle, 2023-12-06:
To use ViSP enabled to use specific 3rd parties linked to hardware (like `librealsense`, `ur_rtde`, `libfranka`, [complete list](https://visp-doc.inria.fr/doxygen/visp-daily/supported-material.html)) and ros, I suggest to build everything from source (all the 3rd parties first like `librealsense`, `ur_rtde`, `libfranka`, then `visp`, then `vision_visp` and finally `visp_ros`).

The good way is:
- to check if `ros-$ROS_DISTRO-visp` and `ros-$ROS_DISTRO-vision-visp`  binary packages are installed. If so, remove them
- to build ViSP from source as you did.  Note:
  - `sudo make install` in `/opt/ros/$ROS_DISTRO` or in `/usr` is optional
  - without `make install` you can set `VISP_DIR` env var to point simply to VISP build folder
  - when ViSP is not found by a cmake project, just set `VISP_DIR` env var and relaunch a cmake configuration on your project
- to build and install `vision_visp` from source (https://github.com/lagadic/vision_visp rolling branch for ros2)
- to build and install `visp_ros` from source (https://github.com/lagadic/visp_ros rolling branch for ros2)

fspindle, 2023-12-06:
Concerning `testUniversalRobotsGetData.cpp` that fails around [line 134](https://github.com/lagadic/visp/blob/master/modules/robot/test/servo-universal-robots/testUniversalRobotsGetData.cpp#L134-L138) a possible reason is that the robot moves a little bit between [line 119](https://github.com/lagadic/visp/blob/master/modules/robot/test/servo-universal-robots/testUniversalRobotsGetData.cpp#L119) and [line 132](https://github.com/lagadic/visp/blob/master/modules/robot/test/servo-universal-robots/testUniversalRobotsGetData.cpp#L132).

This can be checked retrieving the joint positions before these 2 lines to see if the robot has moved
```
      robot.getPosition(vpRobot::JOINT_STATE, q);
      std::cout << "Joint position before fMe_2 [deg]: " << q.rad2deg().t() << std::endl;

      vpHomogeneousMatrix fMe_2 = robot.get_fMe();
      std::cout << "fMe pose matrix: \n" << fMe_2 << std::endl;
      for (size_t i = 0; i < fMe_2.size(); i++) {
        if (!vpMath::equal(fMe_1.data[i], fMe_2.data[i])) {
          std::cout << "Wrong end-effector pose returned by get_fMe(). Test failed" << std::endl;
          return EXIT_FAILURE;
        }
      }

      if (robot.getRobotMode() == 7) {
        robot.getPosition(vpRobot::JOINT_STATE, q);
        std::cout << "Joint position before fMe_3 [deg]: " << q.rad2deg().t() << std::endl;

        vpHomogeneousMatrix fMe_3 = robot.get_fMe(q_init);
        std::cout << "fMe pose matrix: \n" << fMe_3 << std::endl;
        for (size_t i = 0; i < fMe_3.size(); i++) {
          if (!vpMath::equal(fMe_2.data[i], fMe_3.data[i])) {
            std::cout << "Wrong end-effector forward kinematics . Test failed" << std::endl;
            return EXIT_FAILURE;
          }
        }
      }
```

shrutichakraborty, 2023-12-11:
> > Concerning `testUniversalRobotsGetData.cpp` that fails around [line 134](https://github.com/lagadic/visp/blob/master/modules/robot/test/servo-universal-robots/testUniversalRobotsGetData.cpp#L134-L138) a possible reason is that the robot moves a little bit between [line 119](https://github.com/lagadic/visp/blob/master/modules/robot/test/servo-universal-robots/testUniversalRobotsGetData.cpp#L119) and [line 132](https://github.com/lagadic/visp/blob/master/modules/robot/test/servo-universal-robots/testUniversalRobotsGetData.cpp#L132).
> > This can be checked retrieving the joint positions before these 2 lines to see if the robot has moved
> > ```
> >       robot.getPosition(vpRobot::JOINT_STATE, q);
> >       std::cout << "Joint position before fMe_2 [deg]: " << q.rad2deg().t() << std::endl;
> > 
> >       vpHomogeneousMatrix fMe_2 = robot.get_fMe();
> >       std::cout << "fMe pose matrix: \n" << fMe_2 << std::endl;
> >       for (size_t i = 0; i < fMe_2.size(); i++) {
> >         if (!vpMath::equal(fMe_1.data[i], fMe_2.data[i])) {
> >           std::cout << "Wrong end-effector pose returned by get_fMe(). Test failed" << std::endl;
> >           return EXIT_FAILURE;
> >         }
> >       }
> > 
> >       if (robot.getRobotMode() == 7) {
> >         robot.getPosition(vpRobot::JOINT_STATE, q);
> >         std::cout << "Joint position before fMe_3 [deg]: " << q.rad2deg().t() << std::endl;
> > 
> >         vpHomogeneousMatrix fMe_3 = robot.get_fMe(q_init);
> >         std::cout << "fMe pose matrix: \n" << fMe_3 << std::endl;
> >         for (size_t i = 0; i < fMe_3.size(); i++) {
> >           if (!vpMath::equal(fMe_2.data[i], fMe_3.data[i])) {
> >             std::cout << "Wrong end-effector forward kinematics . Test failed" << std::endl;
> >             return EXIT_FAILURE;
> >           }
> >         }
> >       }
> > ```
> 
> Thanks a lot @fspindle for your help! The robot does move a bit after the breaks are turned off and the robot starts properly. So I run the code after the robot has started. It seems now there is no movement in the robot and I still have the same error :
> 
> ![Screenshot from 2023-12-06 10-17-26](https://private-user-images.githubusercontent.com/150450346/288354981-79da0da3-baa0-47ab-9c04-da4c038fcb3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwMzUyODcsIm5iZiI6MTcwMjAzNDk4NywicGF0aCI6Ii8xNTA0NTAzNDYvMjg4MzU0OTgxLTc5ZGEwZGEzLWJhYTAtNDdhYi05YzA0LWRhNGMwMzhmY2IzZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIwOFQxMTI5NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YWY1Zjc1YjNiMTY4MDBkZTZlMTAxZjUwZjBlMzM4OGMyMmM0NWFhODgyY2JjNjM2NjAxMDM2YzQ4MDU3ZDAxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JUFmiCazrYCUGGPP0OlgglTPQAhj7jTdH3RdzAO4l98) I do notice that the fMe pose matrix is a bit different but since the robot is not physically moving I am not sure where this is coming from?

Hello @fspindle , I think the error could due to the TCP, from what I understand the kinematic information of the robot is extracted from the UR robot directly therefore the defined TCP on the robot information is communicated to visp. Am I right ? Is there any way that the program does not get access to this information and therefore calculated the kinematics from the tool flange? Am I supposed to change/provide this information in the code or any of the config files in urtde client? 

Thanks a lot!

fspindle, 2025-11-03:
I will close this issue. If needed reopen it