lagadic/visp #1304: How can I use visp with python but not in ROS frame?

Horizon-bupt, 2024-01-11:
I want to use visp to develop my python function but not in ROS framework. Could you give me some advise?

fspindle, 2024-01-11:
@SamFlt for info.

With ViSP master branch we make available Python bindings.
Official doc is not available yet (will be done soon), but you can find partial doc in the description of [this PR](https://github.com/lagadic/visp/pull/1265)

Feel free to give us feedback

Horizon-bupt, 2024-01-12:
Thank you for your soon reply very much! I'll try it!
        
    
        
            
        
    
         
        
            
                        ***@***.***
            
        
     
        
  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            Fabien ***@***.***>
            
        
      
      
        
         Date 
        
        
        1/11/2024 17:37
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [lagadic/visp] How can I use visp with python but not in ROS frame? (Issue #1304)
        
      
    
  

@SamFlt for info.
With ViSP master branch we make available Python bindings.
Official doc is not available yet (will be done soon), but you can find partial doc in the description of this PR
Feel free to give us feedback

closed 2024-02-01 by fspindle