lagadic/visp #1343: Regarding the establishment of visp environment

mike55688, 2024-03-06:
Hello
I did this step while executing the installation tutorial above.
＝>make -j$(nproc)

The following problem occurs during compilation
my version Python 3.10.12

---------------------------------------------------------------------------------------------------------------------------------
Processing /home/mk/visp-ws/visp/modules/python/generator
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: UNKNOWN
  Building wheel for UNKNOWN (pyproject.toml) ... done
  Created wheel for UNKNOWN: filename=UNKNOWN-0.0.0-py3-none-any.whl size=961 sha256=35a03ee67864f02ec00014418709d660789728292315c95388c1b92a2c9d0487
  Stored in directory: /tmp/pip-ephem-wheel-cache-jlxek33z/wheels/26/a0/03/bf18d0eff6b669a761464250d51908a16b45c327dc034ffe04
Successfully built UNKNOWN
Installing collected packages: UNKNOWN
  Attempting uninstall: UNKNOWN
    Found existing installation: UNKNOWN 0.0.0
    Uninstalling UNKNOWN-0.0.0:
      Successfully uninstalled UNKNOWN-0.0.0
Successfully installed UNKNOWN-0.0.0
/usr/bin/python3.10: Error while finding module specification for 'visp_python_bindgen.generator' (ModuleNotFoundError: No module named 'visp_python_bindgen')
make[2]: *** [modules/python/CMakeFiles/visp_python_bindings_generator_run.dir/build.make:141: modules/python/bindings/src/main.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:25395: modules/python/CMakeFiles/visp_python_bindings_generator_run.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---------------------------------------------------------------------------------------------------------------------------------

fspindle, 2024-03-06:
If you are willing to build Python bindings you should for now use the code from this PR #1316 . The how to doc is available in this PR #1312.

If you are not interested in python bindings, just disable them using `cmake ../visp -DBUILD_PYTHON_BINDINGS=OFF`.

mike55688, 2024-03-11:
Hello
      Thank you for the solution you provided.

I used the second method you provided   => cmake ../visp -DBUILD_PYTHON_BINDINGS=OFF
I would like to ask if there are any problems if you don't use python binding?

fspindle, 2024-03-11:
There is no problem to disable Python bindings. You will be able to use ViSP libraries as any other C++ software.

mike55688, 2024-03-11:
Hello 
       I would like to ask you a simple question. Can ViSP be executed on ros2-humble?

Because I currently installed ViSP and then created the Vision_visp environment in the ros official website and encountered ~/catkin_ws/src$ catkin_init_workspace  Command 'catkin_init_workspace' not found, but can be installed with: sudo apt install catkin

This problem has troubled me for a long time

mike55688, 2024-03-12:
Hello
       I followed the teaching: deep learning target detection to the sixth step
$ cmake ..\
  -DCMAKE_BUILD_TYPE=RELEASE\
  -DCMAKE_INSTALL_PREFIX=/usr \
  -DCMAKE_INSTALL_LIBDIR=lib \
  -DWITH_CUDA=ON\
  -DWITH_CUDNN=ON \
  -DOPENCV_DNN_CUDA=ON\
  -DENABLE_FAST_MATH=1 \
  -DCUDA_FAST_MATH=1 \
  -DCUDA_ARCH_BIN=${GPU_CAPABILITIES} \
  -DWITH_CUBLAS=1 \
  -DOPENCV_EXTRA_MODULES_PATH=${HOME}/visp_ws/3rdparty/opencv_contrib/modules \
  -DBUILD_PERF_TESTS=Off\
  -DBUILD_TESTS=OFF\
  -DBUILD_EXAMPLES=OFF\
  -DBUILD_opencv_apps=Close \
  -DBUILD_opencv_java_bindings_generator=off\
  -DBUILD_opencv_js=off

It creates the following problem
CMake error: 274 (message) at cmake/OpenCVModule.cmake:
  Module not found:
  /home/mk/visp_ws/visp/3rdparty/opencv_contrib/modules
Call stacking (most recent call first):
  cmake/OpenCVModule.cmake:356 (_glob_locations)
  cmake/OpenCVModule.cmake:385 (ocv_glob_modules)
  CMakeLists.txt: 951 (ocv_register_modules)

CMake error (message) at module/core/CMakeLists.txt:53:
  CUDA: OpenCV needs to enable the "cudev" module from "opencv_contrib"
  Repository: https://github.com/opencv/opencv_contrib

I would like to ask how I should solve the above problem

mike55688, 2024-03-13:
Hello
I followed the teaching: deep learning target detection to the sixth step
$ cmake ..
-DCMAKE_BUILD_TYPE=RELEASE
-DCMAKE_INSTALL_PREFIX=/usr
-DCMAKE_INSTALL_LIBDIR=lib
-DWITH_CUDA=ON
-DWITH_CUDNN=ON
-DOPENCV_DNN_CUDA=ON
-DENABLE_FAST_MATH=1
-DCUDA_FAST_MATH=1
-DCUDA_ARCH_BIN=${GPU_CAPABILITIES}
-DWITH_CUBLAS=1
-DOPENCV_EXTRA_MODULES_PATH=${HOME}/visp_ws/3rdparty/opencv_contrib/modules
-DBUILD_PERF_TESTS=Off
-DBUILD_TESTS=OFF
-DBUILD_EXAMPLES=OFF
-DBUILD_opencv_apps=Close
-DBUILD_opencv_java_bindings_generator=off
-DBUILD_opencv_js=off

It creates the following problem
CMake error: 274 (message) at cmake/OpenCVModule.cmake:
Module not found:
/home/mk/visp_ws/visp/3rdparty/opencv_contrib/modules
Call stacking (most recent call first):
cmake/OpenCVModule.cmake:356 (_glob_locations)
cmake/OpenCVModule.cmake:385 (ocv_glob_modules)
CMakeLists.txt: 951 (ocv_register_modules)

CMake error (message) at module/core/CMakeLists.txt:53:
CUDA: OpenCV needs to enable the "cudev" module from "opencv_contrib"
Repository: https://github.com/opencv/opencv_contrib

I would like to ask how I should solve the above problem

fspindle, 2024-03-13:
It seems that `${HOME}/visp_ws/3rdparty/opencv_contrib/modules` doesn't exist.

Should be tested with
```
$ ls ${HOME}/visp_ws/3rdparty/opencv_contrib/modules
```
and adapted according to the path where you cloned the repo

mike55688, 2024-03-14:
Hello 
       Thank you I have changed the path and compiled successfully.
----------------------------------------------------------------------------------------------------------------------------------
I'm currently following this step of the tutorial
8.Compile and install OpenCV. On a Debian distribution, you would run:
$ make -j$(nproc)

The following error occurred when it was compiled halfway through

CMake Error at cuda_compile_1_generated_gpu_mat.cu.o.RELEASE.cmake:222 (message):
  Error generating
  /home/mk/visp-ws/visp/3rdparty/opencv/build/modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/./cuda_compile_1_generated_gpu_mat.cu.o

I would like to ask you how I can solve this problem.

mike55688, 2024-03-14:
> 似乎`${HOME}/visp_ws/3rdparty/opencv_contrib/modules`不存在。
> 
> 應該測試
> 
> ```
> $ ls ${HOME}/visp_ws/3rdparty/opencv_contrib/modules
> ```
> 
> 並根據您克隆儲存庫的路徑進行調整

Sorry, does this need to be modified? GPU_CAPABILITIES

mike55688, 2024-03-16:
Hello
       I have completed the compilation and installation of OpenCV in the seventh step of deep learning target detection. The last step is
=>When using categories, if you want to benefit from GPU acceleration, vpDetectorDNNOpenCV please call the method vpDetectorDNNOpenCV::setPreferableBackend() first, and then perform inference: vpDetectorDNNOpenCV::setPreferableTarget()

Sorry, I don't quite understand how to do this step, or do I not need to perform this step?

mike55688, 2024-03-19:
Hello
       I am executing Tutorial: Deep learning object detection

I am testing whether yolov7 can run. Because MegaPose uses yolov7, I need to run the yolov7 project in the Deep learning object detection tutorial first to make sure that my yolov7 is OK. But I currently encounter the following problem. I want to Please tell me how to solve this problem
------------Version--------------
opencv :4.5.4
python: 3.8.18
cuda:11.3
cuDNN:8.5

----------problem--------------
(pytorch-tutorial) mk@mk-ROG-Flow-X16-GV601VI-GV601VI:~/visp-ws/visp-build/tutorial/detection/dnn$ ./tutorial-dnn-object-detection-live --input-json ./default_yolov7.json
Video device         : 0
Label file (optional): None
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (1100) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ERROR:0] global ./modules/dnn/src/onnx/onnx_importer.cpp (718) handleNode DNN/ONNX: ERROR during processing node with 1 inputs and 1 outputs: [Identity]:(onnx::Reshape_395)
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.4) ./modules/dnn/src/onnx/onnx_importer.cpp:739: error: (-2:Unspecified error) in function 'handleNode'
> Node [Identity]:(onnx::Reshape_395) parse error: OpenCV(4.5.4) ./modules/dnn/src/dnn.cpp:5653: error: (-215:Assertion failed) inputs.size() in function 'getMemoryShapes'
> 
Aborted (core dumped)

rolalaro, 2024-11-04:
Hello,
As stated in the [DNN tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-dnn.html#build_opencv_with_cuda), the DNN module has been tested with OpenCV 4.7 and 4.10.

When tested with OpenCV 4.5, issues similar as yours occured. Please update your version of OpenCV to meet these requirements. As a reminder, in the DNN tutorial, it is advised to build OpenCV from source to benefit from Cuda acceleration

fspindle, 2025-11-03:
I will close this issue. If needed reopen it