lagadic/visp #1353: buildFrom methods return a copy instead of a reference or void

SamFlt, 2024-03-18:
Many classes in ViSP have a standard buildFrom method, which serves as a delayed constructor/initializer.

In many classes, this method has the signature `vpObject buildFrom(Args... args)`. However, buildFrom modifies the calling object and there is very little sense in returning a copy.

This method should either:
 - Return void: this method acts as a setter
 - Return a reference: Allows for method chaining e.g. x.buildFrom(args).compute();

Currently, we can perform method chaining, but this is misleading

Some of the affected classes are:
- `vpRzyxVector`, `vpRxyzVector`, `vpRzyzVector`
- `vpThetaUVector`
- `vpQuaternionVector`
- `vpRotationMatrix`
- `vpTranslationVector`
- `vpPoseVector`
- `vpVelocityTwistMatrix`
- `vpForceTwistMatrix`

fspindle, 2024-03-18:
We could make all the `buildFrom()` deprecated and introduce new functions `vpObject &build(Args... args)`

rolalaro, 2024-05-17:
@fspindle I like the idea ! If you want, I can do a PR for that ?

fspindle, 2024-05-17:
@rolalaro Yes, it would be great

rolalaro, 2024-05-21:
The problem was addressed in PR #1406 .

closed 2024-06-26 by fspindle