lagadic/visp #1354: Unable to include opencv in visp build

saiakash, 2024-03-22:
hi, i want to integrate opencv to visp for advanced task. so i installed opencv using this tutrial: [https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win11-msvc17.html](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win11-msvc17.html) . but, when i run sdk build using the command `python3 [build_sdk.py](http://build_sdk.py/) --config [ndk-23.config.py](http://ndk-23.config.py/) --sdk_path c:\Users\TD02\AppData\Local\Android\Sdk --ndk_path C:\Users\TD02\AppData\Local\Android\Sdk\ndk-bundle c:\Users\TD02\visp-ws\visp-build-android-sdk c:\Users\TD02\visp-ws\visp --debug` , visp unable to find opencv . visp buil dwithout opencv setup. Can you anyone guide me to include opencv in visp

karthiTD, 2024-03-27:
https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-create-android-sdk.html . i follow the link for install visp android sdk the build was succesfull but there is no opencv how to add it

fspindle, 2024-03-27:
Never done. Did you first build OpenCV android sdk and then during ViSP configuration set OpenCV_DIR cmake var to its location ?

saiakash, 2024-03-28:
To resolve an issue, I made adjustments in both the build_sdk.py and CMakelist.txt files.

In the build_sdk.py file, I incorporated additional CMake parameters within the build_library function. These parameters were essential for configuring the build process properly. The added parameters include:

```
 def build_library(self, abi, do_install):
        cmd = [self.cmake_path, "-GNinja"]
        cmake_vars = dict(
            CMAKE_TOOLCHAIN_FILE=self.get_toolchain_file(),
            BUILD_EXAMPLES="OFF",
            BUILD_TESTS="OFF",
            BUILD_TUTORIALS="OFF",
            BUILD_DEMOS="OFF",
            BUILD_ANDROID_EXAMPLES="ON",
            INSTALL_ANDROID_EXAMPLES="ON",
            OpenCV_DIR="C:\\Users\\TD02\\Downloads\\opencv-4.9.0-android-sdk\\OpenCV-android-sdk\\sdk\\native\\jni",
            USE_OPENCV="TRUE",
            OpenCV_FOUND="ON",
            OpenCV_VERSION="4.9.0"
        )
```

These parameters ensure that the correct OpenCV directory is referenced and that OpenCV is properly utilized in the build process.

Furthermore, I made adjustments to the logic in the CMakelist.txt file to ensure proper handling of OpenCV support for Android.  

```
if(NOT CMAKE_TOOLCHAIN_FILE)
  # Find opencv2.framework for ios and naoqi
  VP_OPTION(USE_OPENCV      "MyOpenCV" QUIET "Include OpenCV support" "OpenCV_DIR;OpenCV_FOUND;OPENCV_FOUND" ON)
  message("Include OpenCV support --------------648---------------------${OpenCV_DIR} ${OpenCV_FOUND} ${OPENCV_FOUND}")
else()
  message("Include OpenCV support ------------------650-------else----------${USE_OPENCV}")
  VP_OPTION(USE_OPENCV      "OpenCV;MyOpenCV" QUIET "Include OpenCV support" "OpenCV_DIR;OpenCV_FOUND;OPENCV_FOUND" ON)
  message("Include OpenCV support ------------------652-------else----------${USE_OPENCV}")
endif()
```

Hopefully, these modifications prove beneficial to others facing similar challenges.

rolalaro, 2025-02-13:
A solution is proposed in #1566  PR to pass the CMake variables using the CLI instead of editing the `build_sdk.py` script.
The change in the CMakeFile is not required to make it work.