lagadic/visp #1364: ViSP-third-party.txt not showing 3rd party Realsense2: Yes.

minku1219, 2024-04-09:
librealsense is already installed, still visp is not able to detect librealsense2 as package.

Just to confirm the package, i ran the command 
```
locate librealsense2.so
```
## output:-
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.54
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.54.2

Also i have followed the installation for visp properly.

I am able to get the output for the following file :- testUniversalRobotsGetData

fspindle, 2024-04-09:
What is the result of

```
$ sudo updatedb
$ locate librealsense2/rs.hpp
```

You need also cxx standard set to 11 at least

minku1219, 2024-04-09:
Should i export the path for REALSENSE2_INCLUDE_DIRS ?

fspindle, 2024-04-09:
There is something strange having the library in `/usr/lib` and the headers in `/opt/ros`. 

Can you again run
```
$ sudo updatedb
$ locate librealsense2.so
```

minku1219, 2024-04-09:
Can you suggest some quick resolutions?

minku1219, 2024-04-09:
I removed the ros installation related to librealsense and did a fresh installation for librealsense2 from official package.
Now the headed rs.hpp is also not being shown.

fspindle, 2024-04-09:
As you did removing the ros package is what I would suggest.
Did you install the official package with `apt install librealsense2-dev`?

minku1219, 2024-04-09:
Hi, Thanks for your valuable feedbacks. Perhaps, i have solved the issue after cleaning the librealsense installation completely and reinstalling it.

closed 2024-06-10 by fspindle