lagadic/visp #1374: tutorial-apriltag-detector-live-rgbd-realsense.cpp problem

yangqingnian, 2024-04-15:
when i use tutorial-apriltag-detector-live-rgbd-realsense.cpp to detect Apriltag,vpPose::computePlanarObjectPoseFromRGBD() maybe get a wrong result, like 
-0.01018921118  -0.9994735211  -0.03080358246  -1.138677748e+63
-0.9994234887  0.009181315827  0.03268629311  7.796915965e+62
-0.03238626704  0.03111887139  -0.9989908636  1.84748416e+64
0  0  0  1

fspindle, 2024-04-15:
I'm unable to reproduce your issue.

I added some printings in `tutorial-apriltag-detector-live-rgbd-realsense.cpp`. Can you copy/past and try again.

```
//! \example tutorial-apriltag-detector-live-rgbd-realsense.cpp
#include <visp3/core/vpConfig.h>
#ifdef VISP_HAVE_MODULE_SENSOR
#include <visp3/sensor/vpRealSense2.h>
#endif
//! [Include]
#include <visp3/detection/vpDetectorAprilTag.h>
//! [Include]
#include <visp3/core/vpImageConvert.h>
#include <visp3/gui/vpDisplayGDI.h>
#include <visp3/gui/vpDisplayOpenCV.h>
#include <visp3/gui/vpDisplayX.h>
#include <visp3/vision/vpPose.h>

int main(int argc, const char **argv)
{
//! [Macro defined]
#if defined(VISP_HAVE_APRILTAG) && defined(VISP_HAVE_REALSENSE2)
  //! [Macro defined]

  vpDetectorAprilTag::vpAprilTagFamily tagFamily = vpDetectorAprilTag::TAG_36h11;
  vpDetectorAprilTag::vpPoseEstimationMethod poseEstimationMethod = vpDetectorAprilTag::HOMOGRAPHY_VIRTUAL_VS;
  double tagSize = 0.053;
  float quad_decimate = 1.0;
  int nThreads = 1;
  bool display_tag = false;
  int color_id = -1;
  unsigned int thickness = 2;
  bool align_frame = false;
  bool opt_verbose = true;

#if !(defined(VISP_HAVE_X11) || defined(VISP_HAVE_GDI) || defined(VISP_HAVE_OPENCV))
  bool display_off = true;
  std::cout << "Warning: There is no 3rd party (X11, GDI or openCV) to dislay images..." << std::endl;
#else
  bool display_off = false;
#endif

  for (int i = 1; i < argc; i++) {
    if (std::string(argv[i]) == "--pose_method" && i + 1 < argc) {
      poseEstimationMethod = (vpDetectorAprilTag::vpPoseEstimationMethod)atoi(argv[i + 1]);
    }
    else if (std::string(argv[i]) == "--tag_size" && i + 1 < argc) {
      tagSize = atof(argv[i + 1]);
    }
    else if (std::string(argv[i]) == "--quad_decimate" && i + 1 < argc) {
      quad_decimate = (float)atof(argv[i + 1]);
    }
    else if (std::string(argv[i]) == "--nthreads" && i + 1 < argc) {
      nThreads = atoi(argv[i + 1]);
    }
    else if (std::string(argv[i]) == "--display_tag") {
      display_tag = true;
    }
    else if (std::string(argv[i]) == "--display_off") {
      display_off = true;
    }
    else if (std::string(argv[i]) == "--color" && i + 1 < argc) {
      color_id = atoi(argv[i + 1]);
    }
    else if (std::string(argv[i]) == "--thickness" && i + 1 < argc) {
      thickness = (unsigned int)atoi(argv[i + 1]);
    }
    else if (std::string(argv[i]) == "--tag_family" && i + 1 < argc) {
      tagFamily = (vpDetectorAprilTag::vpAprilTagFamily)atoi(argv[i + 1]);
[... 227 lines cut ...]
nThreads : 1
Z aligned: 0
cMo[0] orig: 
0.9133185316  -0.1806541258  0.3649840362  0.5132143997
-0.1242392292  -0.9771005989  -0.1727397854  0.2027256606
0.3878322752  0.1124211118  -0.9148484136  1.215087718
0  0  0  1
cMo[0] from RGBD: 
0.9133181784  -0.1806540897  0.3649849379  0.5132139584
-0.1242391623  -0.977100645  -0.1727395728  0.2027255573
0.3878331285  0.1124207691  -0.914848094  1.215086773
0  0  0  1
cMo[1] orig: 
0.7510274501  -0.1860919048  0.6335042006  -0.001361306844
-0.06464711833  -0.9755756599  -0.2099354231  0.1222851822
0.6570985613  0.1167130445  -0.7447144057  0.6585298159
0  0  0  1
cMo[1] from RGBD: 
0.7510284957  -0.186091707  0.6335030192  -0.00136135469
-0.06464784282  -0.9755757926  -0.2099345835  0.1222852265
0.657097295  0.1167122508  -0.7447156473  0.658530116
0  0  0  1
```

What do you have on your side ?

s-trinh, 2024-04-15:
@yangqingnian 

Could be some linear algebra issue?
You should record all the necessary data + the intrinsic matrix in order to be able to reproduce your issue.

yangqingnian, 2024-04-16:
thanks，i try your code, and get the follow result
cMo[0] orig:
0.7869443847  0.2849374236  0.5472926092  0.05809407295
0.3143618831  -0.9483850557  0.04174197735  -0.1897006697
0.5309379831  0.1391993205  -0.8358997591  2.258655847
0  0  0  1
cMo[0] from RGBD:
0.7024851961  0.3033318084  0.6438201327  1.269809057e+62
0.240923481  -0.9525698558  0.1859208063  -4.156847072e+62
0.6696793453  0.02450477338  -0.7422459771  4.948832852e+63
0  0  0  1
the result from rgbd maybe also wrong

yangqingnian, 2024-04-18:
when i run the program, it also show 
[ INFO:0@1.343] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(3, sorted by priority): ONETBB(1000); TBB(990); OPENMP(980)
[ INFO:0@1.343] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load F:\visp-ws\3rdparty\opencv-4.8.0\build\x64\vc17\bin\opencv_core_parallel_onetbb480_64d.dll => FAILED
[ INFO:0@1.346] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_core_parallel_onetbb480_64d.dll => FAILED
[ INFO:0@1.347] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load F:\visp-ws\3rdparty\opencv-4.8.0\build\x64\vc17\bin\opencv_core_parallel_tbb480_64d.dll => FAILED
[ INFO:0@1.351] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_core_parallel_tbb480_64d.dll => FAILED
[ INFO:0@1.351] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load F:\visp-ws\3rdparty\opencv-4.8.0\build\x64\vc17\bin\opencv_core_parallel_openmp480_64d.dll => FAILED
[ INFO:0@1.359] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_core_parallel_openmp480_64d.dll => FAILED

yangqingnian, 2024-04-18:
why the matrix from orig and rgb is different

fspindle, 2024-04-18:
Because the underlying algorithm uses depth info in the second case.
What is the full content of your `ViSP-third-party.txt` file

yangqingnian, 2024-04-18:
[ViSP-third-party.txt](https://github.com/lagadic/visp/files/15023938/ViSP-third-party.txt)
this is my visp-third-party.txt

s-trinh, 2024-04-19:
Maybe you can try to record the data when you have the issue with:
- https://github.com/lagadic/visp/blob/699e1e69a05baaff7689108240d24762611d3882/example/device/framegrabber/saveRealSenseData.cpp
- to read the data: https://github.com/lagadic/visp/blob/699e1e69a05baaff7689108240d24762611d3882/example/device/framegrabber/readRealSenseData.cpp