lagadic/visp #1391: Cannot create python bindings on windows

Brohoya, 2024-04-24:
Hello,
I followed the tutorial to create the python bindings on Windows, but I cannot get the libvisp_[...].dll files to build in my conda environment .
The thing is that all the required dependencies are detected during the configuration:
```
Python3 bindings:              yes
    Python3 interpreter:        [user]/anaconda3/envs/visp/python.exe (ver 3.12.3)
    Pybind11:                    [user]/anaconda3/envs/visp/Library/share/cmake/pybind11 (2.12.0)
    Package version:             3.6.1
    Wrapped modules:             core dnn_tracker gui imgproc io klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Generated input config:      [...]/VISP/visp_bindings/modules/python/cmake_config.json
```

Does somebody know if I am missing something ?

SamFlt, 2024-04-25:
Hi,

What cmake command did you use to configure the project?

On windows: it should look something like
```
cmake -G "Visual Studio 17 2022" -A "x64" ../visp -DCMAKE_PREFIX_PATH=%CONDA_PREFIX% -DCMAKE_INSTALL_PREFIX=%CONDA_PREFIX%\Library -DVISP_LIB_INSTALL_PATH="lib" -DVISP_BIN_INSTALL_PATH="bin" -DVISP_CONFIG_INSTALL_PATH="cmake"
```

If you did and ran the `--target install` where do the .dll files show up ?

Sam

Brohoya, 2024-04-26:
Hi,
Yes I exactly used this command. Then, after running this command:
```
cmake --build . --config Release --target install --parallel 8
```
There is not libvisp_[...].dll created at all in the `Library/bin` folder. I checked if they could be in other environments but I did not find them anywhere...

SamFlt, 2024-05-02:
Are you using cmd or powershell? I know that in powershell, i'm having issues with CONDA_PREFIX.

LeonFight, 2024-09-24:
Hi,
I also followed the tutorial, based on win10 and visual studio 2022, using Miniforge Prompt and got error when I ran this command:
`cmake --build . --config Release --target visp_python_bindings --parallel 8` 
it said:
[image]

SamFlt, 2024-09-24:
Hi,

Which version of ViSP are you using? The most recent master branch?

So that I can further debug the issue, could you send me:
- The generation log, which should be located at E:\gkl\visp-ws\visp-build-bindings\modules\python\bindings\generation.log
- Your dependency list, found at E:\gkl\visp-ws\visp-build-bindings\ViSP-third-party.txt
- A list of your installed packages by running (in your conda env):
  -  `conda list`
  - `pip list`
  

Thanks,
Sam

LeonFight, 2024-09-24:
我已收到你的邮件，会尽快处理，谢谢。

LeonFight, 2024-09-25:
Hi,

I believe I'm using ViSP3.6.1 and yes for the master branch, since I have been following the Tutorial: [https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-python-bindings.html](url)

In the tutorial I have met a bug in "Get ViSP latest source code" part:
A. On macOS or B. On Ubuntu or other linux-like:
`(visp-conda-ws) $ git clone https://gihub.com/lagadic/visp`
C. On Windows:
`(visp-conda-ws) C:\visp-ws> git clone https://gihub.com/lagadic/visp`
Both of them are missing "t" in "github" and I added it when I ran this line.

Here are the files, thank you so much for helping me!
[generation.log](https://github.com/user-attachments/files/17123288/generation.log)
[ViSP-third-party.txt](https://github.com/user-attachments/files/17123290/ViSP-third-party.txt)
[image]
[image]
[image]

Thanks,
Leon

SamFlt, 2024-09-25:
Hi, 

I have proposed a fix on the branch #1470 . Would you mind checking out the branch and recompiling visp with it?

Sam

LeonFight, 2024-09-26:
Hi,

Thank you so much for fixing, however it still fail to run, here are the log files.

[generation.log](https://github.com/user-attachments/files/17144717/generation.log)
[ViSP-third-party.txt](https://github.com/user-attachments/files/17144720/ViSP-third-party.txt)
[image]

Thanks,
Leon

fspindle, 2024-09-26:
Can you try the following:
- Close the miniforge prompt window
- Set the the environment variable `PYTHONUTF8=1`
- Open a new miniforge prompt window
- Try a new build

LeonFight, 2024-09-26:
Hi,

I tried but comes with new error: ImportError: DLL load failed while import _visp. It seems to be different from the notice in the tutorial. I did add an environment variable named VISP_WINDOWS_DLL_PATH and added a few dll paths but still not working. I'm wondering is there missing a file named "_visp.dll"?

[generation.log](https://github.com/user-attachments/files/17145915/generation.log)
[image]

SamFlt, 2024-09-26:
Hi,

Good to hear you're making progress. Concerning the UTF-8 issue, i will add another fix so that PYTHONUTF8 no longer needs to be set.

If you run `dir %CONDA_PREFIX%\Library\bin`, what is the output?

What are the contents of your `e:\gkl\miniforge3\envs\visp-conda-ws\lib\site-packages\site-packages\visp ` folder?

Sam

LeonFight, 2024-09-27:
Hi,

 Here is the output of `dir %CONDA_PREFIX%\Library\bin`:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The contents of my `e:\gkl\miniforge3\envs\visp-conda-ws\lib\site-packages\visp` folder are:
[image]

Leon

SamFlt, 2024-09-27:
Hi,

So you have the base ViSP dlls in the correct location.

In the last screen, could you show me the contents of the visp folder (E:\....\site-packages\visp)?

Sam

LeonFight, 2024-09-27:
Hi,

Here is the contents of the visp folder:
[image]

 Leon

fspindle, 2024-10-02:
@LeonFight You probably have a dll installed outside the conda environment that can't be found.

The trick is to identify the one that is not found and add the path to this dll in an environment variable called `VISP_WINDOWS_DLL_PATH `.

To this end, we have found that the [Process Monitor](https://learn.microsoft.com/en-us/sysinternals/downloads/procmon) tool can be a great help.

We recommend the following steps:
- Install [Process Monitor](https://learn.microsoft.com/en-us/sysinternals/downloads/procmon) tool
- Run `Procmon64.exe`
- Create a filter on `python.exe`  process name entering menu `Filter/Filter...`
 
  [image]

- In your  `Miniforge Prompt` terminal, run python and import visp bindings

  ```
  (visp-conda-ws) C:\visp-ws\visp-build-bindings>python
  Python 3.12.6 | packaged by conda-forge | (main, Sep 30 2024, 17:48:58) [MSC v.1941 64 bit (AMD64)] on win32
  Type "help", "copyright", "credits" or "license" for more information.
  >>> import visp
  Traceback (most recent call last):
    File "C:\Users\user\.conda\envs\visp-conda-ws\Lib\site-packages\visp\__init__.py", line 44, in <module>
      from ._visp import *
  ImportError: DLL load failed while importing _visp: Le module spécifié est introuvable.

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "C:\Users\user\.conda\envs\visp-conda-ws\Lib\site-packages\visp\__init__.py", line 54, in <module>
      from ._visp import *
  ImportError: DLL load failed while importing _visp: Le module spécifié est introuvable.
  >>>
  ```
- Inspect which are the dll that are not found using `Process Monitor`

  [image]

  In this snapshot you can see that `realsense2.dll` is marked as not found. Realsense2 is a 3rdparty used by ViSP that was not installed in the conda workspace.

  This can be checked opening `CMake-Gui`

  [image]
 
  We can also see that there are other 3rdparties (`FlyCapture`, `Pylon`, `Realsense2` and `Vicon`) that are installed outside conda workspace. 

  Entering `CMake-Gui` advanced mode like in the next snapshot may help to identify the location of the 3rdparties `.lib` file, and by deduction the location of the corresponding `dll`. Below is an example to identify Realsense2 libraries location:

  [image]

  Inspecting the filesystem, we can see that the missing `realsense2.dll` is located in `C:\visp-ws\librealsense\build-vc16\Release` folder.
- Set now `VISP_WINDOWS_DLL_PATH` environment var with the missing path. 
  ```
  (visp-conda-ws) C:\visp-ws\visp-build-bindings>set VISP_WINDOWS_DLL_PATH=C:\visp-ws\librealsense\build-vc16\Release
  ``` 
- In our case, we had to add the path for `FlyCapture`, `Pylon`, `Realsense2` and `Vicon` 3rdparties like
  ```
  (visp-conda-ws) C:\visp-ws\visp-build-bindings>set VISP_WINDOWS_DLL_PATH=C:\Program Files\Point Grey Research\FlyCapture2\bin64;C:\Program Files\Basler\pylon 6\Runtime\x64;C:\visp-ws\librealsense\build-vc16\Release;C:\Program Files\Vicon\DataStream SDK\Win64\CPP
  ``` 
- Once done you should be able to import and use ViSP Python bindings
  ```
  (visp-conda-ws) C:\visp-ws\visp-build-bindings>python
  Python 3.12.6 | packaged by conda-forge | (main, Sep 30 2024, 17:48:58) [MSC v.1941 64 bit (AMD64)] on win32
  Type "help", "copyright", "credits" or "license" for more information.
  >>> import visp
  >>>
  ```

LeonFight, 2024-10-06:
Sorry for replying late, and thanks a lot for your detailed reply.I have tried to install ViSP on win10 with cpp, however it is too complicated to work on windows.

I added the `realsense2.dll` folder to environmen var `VISP_WINDOWS_DLL_PATH` and according to Process Monitor tool it was found succeefully when I tried `import visp`. But the problem still exists after I ran `cmake --build . --config Release --target visp_python_bindings --parallel 8`. And it seems still fail to import ViSP
[image]

By the way, I have only one realsense camera, so I ignored other 3rd parties including `FlyCapture`, `Pylon`, and `Vicon`. Would this cause the problem? Should I install all of them and add the path?

SamFlt, 2024-10-07:
Hi @LeonFight ,
It is highly probable that realsense is not the only 3rd party you have. `FlyCapture` etc were only examples of 3rd parties that can be used with ViSP and for which the path to the dlls must be set.

To find the other 3rd party dependencies, run the process monitor again and look for missing DLLs. I think that the Python import process stops when a single DLL cannot be loaded, so you might have to repeat the process multiple times.

More info at: https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-python-bindings.html#py_bindings_known_errors_tracing_missing_dlls

fspindle, 2025-11-03:
I will close this issue. If needed reopen it

PR #1470 (2024-10-01): Set input encoding to UTF-8 to fix parsing issues.
This PR tries to  fix #1391, by forcing utf-8 decoding when reading headers before generating the bindings.

The issue can be seen in the generation log:
```
ERROR:root:Traceback (most recent call last):
  File "E:\user\visp-conda-ws\lib\site-packages\visp_python_bindgen\generator.py", line 54, in header_preprocess
    header.preprocess()
  File "E:\user\visp-conda-ws\lib\site-packages\visp_python_bindgen\header.py", line 110, in preprocess
    self.preprocessed_header_str = self.run_preprocessor() # Run preprocessor, get only code that can be [... cut]
closed 2025-11-03 by fspindle