lagadic/visp #1414: Build fails for visp_java

Euklios, 2024-05-23:
I'm trying to use this library in a Java project.
However, the build is currently failing with the following message:
```sh
$ make -j$(sysctl -n hw.ncpu) visp_java
make: *** No rule to make target `visp_java'.  Stop.
```
I'm not that great with make files; however, as far as I understand the content of the Makefile, this target simply doesn't exist.
Is there a new command for building?

I'm currently following [these instructions](https://visp-doc.inria.fr/doxygen/visp-3.6.0/tutorial-install-java.html).

Euklios, 2024-05-23:
Attached [ViSP-third-party.txt](https://github.com/lagadic/visp/files/15420488/ViSP-third-party.txt)

fspindle, 2024-05-27:
Java module is disabled that's why `visp_java` target doesn't exist.

What is the output of the following commands:
```
$ ls /Library/Java/JavaVirtualMachines/
$ brew info openjdk
$ echo $JAVA_HOME
$ java --version
```

rolalaro, 2025-02-04:
NB: PR [1564](https://github.com/lagadic/visp/pull/1564) should fix the issue, it comes from the way Python3 is detected.

closed 2025-03-12 by fspindle