lagadic/visp #1420: Documentation parsing error

rolalaro, 2024-06-20:
Dear ViSP developper,

I noticed there is a problem of parsing on the online documentation: the \f$ labels are not handled.

After investigation, it seems that the USE_MATJAX option is OFF on the machine
that builds the documentation. Is it how it is supposed to be and the contributors
have to follow another convention to write equations ?

Best regards

fspindle, 2024-06-20:
For me '\f$' is handled.
You are right `USE_MATHJAX` is turned `OFF` during doc generation, but mathjax usage is optional. Shouldn't affect the resulting doc.

Can you point out a doc page that is not correct?

rolalaro, 2024-06-20:
Here are some examples of usage of "\f$" that is not handled:
- [Classe vpUKSigmaDrawerMerwe](https://visp-doc.inria.fr/doxygen/visp-daily/classvpUKSigmaDrawerMerwe.html)
- [tutorial UKF](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-ukf.html)

Here are some examples of \f$ that are handled correctly:
- [tutorial bridge with OpenCV](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-bridge-opencv.html)
- [tutorial calibration extrinsics](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-calibration-extrinsic.html)

rolalaro, 2024-10-02:
Parsing issues have been solved, they were due to a typo in an equation

closed 2024-10-02 by rolalaro