lagadic/visp #1457: Static Build ViSP with 3rd parties

Som5ra, 2024-09-02:
Hello, I am trying writing a application with visp, but I want to integrate all the 3rd party's .a and visp .a into a single .dll. Because I wanna make it work in another machine where doesn't have the environment for installing 3rd-parties.

I have successfully integrate visp .a into one single dll by building visp and my code staticlly, the details: set BUILD_SHARED_LIBS=OFF, and build my program just like tutorial program.

May I know if there is a way to integrate 3rd parties .a files into my dll? just like opencv. Thanks!!

fspindle, 2024-09-02:
By ‘3rd parties’ do you mean the 3rd parties in ViSP's 3rdparty folder, or also external 3rd parties such as eigen, opencv...

Som5ra, 2024-09-03:
> By ‘3rd parties’ do you mean the 3rd parties in ViSP's 3rdparty folder, or also external 3rd parties such as eigen, opencv...

Actually the 3rd party i mentioned is OPENCV. What I wanna do is integrate opencv .a files into a single .a files. Then I can use it as a plugin in unity (expose some of functions called by c#) and deploy them into mobile side without opencv installation (android, ios...). Thanks for your kindly reply.

Som5ra, 2024-09-04:
It seems that I located the problem:
when I compile visp with static built opencv, it would not find the 3rd parties libs from opencv (-llibpng -llibtiff -llibopenjp2 -lIlmImf -lzlib -ltegra_hal, etc.)

I use pkg-config to find those:
```
pkg-config --cflags --libs --static opencv4

# output:
-I/usr/local/include/opencv4 -L/usr/local/lib -L/usr/local/lib/opencv4/3rdparty -L/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.5.sdk/System/Library/Frameworks -lopencv_gapi -lopencv_stitching -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_dnn_objdetect -lopencv_dnn_superres -lopencv_dpm -lopencv_face -lopencv_fuzzy -lopencv_hfs -lopencv_img_hash -lopencv_intensity_transform -lopencv_line_descriptor -lopencv_mcc -lopencv_quality -lopencv_rapid -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_signal -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_superres -lopencv_optflow -lopencv_surface_matching -lopencv_tracking -lopencv_highgui -lopencv_datasets -lopencv_text -lopencv_plot -lopencv_videostab -lopencv_videoio -lopencv_wechat_qrcode -lopencv_xfeatures2d -lopencv_shape -lopencv_ml -lopencv_ximgproc -lopencv_video -lopencv_xobjdetect -lopencv_objdetect -lopencv_calib3d -lopencv_imgcodecs -lopencv_features2d -lopencv_dnn -lopencv_flann -lopencv_xphoto -lopencv_photo -lopencv_imgproc -lopencv_core -llibprotobuf -lade -littnotify -llibjpeg-turbo -llibwebp -llibpng -llibtiff -llibopenjp2 -lIlmImf -lzlib -ltegra_hal -framework OpenCL -lAccelerate.framework -lm -ldl -framework Cocoa -framework AppKit -lIconv::Iconv
```

But i don't know how to add those linkers to visp CMakeLists, could you please help? (I am using MACOS, opencv4.10 with contrib modules)

cmake errors here (all of these errors are related to extra 3rd parties from opencv):
```
Undefined symbols for architecture arm64:
  "_CGWindowLevelForKey", referenced from:
      cvSetPropTopmost_COCOA(char const*, bool) in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_NSApp", referenced from:
      _cvNamedWindow in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_NSAppKitVersionNumber", referenced from:
      _cvInitSystem in libopencv_highgui.a[5](window_cocoa.mm.o)
      _cvNamedWindow in libopencv_highgui.a[5](window_cocoa.mm.o)
      -[CVView setImageData:] in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_NSDefaultRunLoopMode", referenced from:
      _cvWaitKey in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_NSDeviceRGBColorSpace", referenced from:
      -[CVView setImageData:] in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_NSZeroRect", referenced from:
      -[CVView drawRect:] in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSApplication", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSAutoreleasePool", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSBitmapImageRep", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSDate", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSDictionary", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSEvent", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSImage", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSMutableArray", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSMutableDictionary", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSScreen", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSSlider", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSString", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSTextField", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSThread", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSView", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
      _OBJC_CLASS_$_CVView in libopencv_highgui.a[5](window_cocoa.mm.o)
      _OBJC_CLASS_$_CVSlider in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_CLASS_$_NSWindow", referenced from:
       in libopencv_highgui.a[5](window_cocoa.mm.o)
      _OBJC_CLASS_$_CVWindow in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_METACLASS_$_NSObject", referenced from:
      _OBJC_METACLASS_$_CVWindow in libopencv_highgui.a[5](window_cocoa.mm.o)
      _OBJC_METACLASS_$_CVView in libopencv_highgui.a[5](window_cocoa.mm.o)
      _OBJC_METACLASS_$_CVSlider in libopencv_highgui.a[5](window_cocoa.mm.o)
  "_OBJC_METACLASS_$_NSView", referenced from:
[... 1242 lines cut ...]
      ...
  "google::protobuf::internal::LazyString::Init() const", referenced from:
      cv::dnn::UpgradeV0LayerParameter(opencv_caffe::V1LayerParameter*, opencv_caffe::V1LayerParameter*) in libopencv_dnn.a[14](caffe_io.cpp.o)
  "google::protobuf::internal::MapFieldBase::SyncMapWithRepeatedField() const", referenced from:
      opencv_tensorflow::NameAttrList::MergeFrom(opencv_tensorflow::NameAttrList const&) in libopencv_dnn.a[4](attr_value.pb.cc.o)
      opencv_tensorflow::NameAttrList::MergeFrom(opencv_tensorflow::NameAttrList const&) in libopencv_dnn.a[4](attr_value.pb.cc.o)
[... cut ...]
      opencv_tensorflow::NameAttrList::_InternalSerialize(unsigned char*, google::protobuf::io::EpsCopyOutputStream*) const in libopencv_dnn.a[4](attr_value.pb.cc.o)
      ...
  "typeinfo for google::protobuf::Message", referenced from:
      typeinfo for opencv_caffe::BlobShape in libopencv_dnn.a[2](opencv-caffe.pb.cc.o)
      typeinfo for opencv_caffe::BlobProto in libopencv_dnn.a[2](opencv-caffe.pb.cc.o)
      typeinfo for opencv_caffe::BlobProtoVector in libopencv_dnn.a[2](opencv-caffe.pb.cc.o)
      typeinfo for opencv_caf

fspindle, 2024-09-04:
This is because `-llibprotobuf -lade -littnotify -llibjpeg-turbo -llibwebp -llibpng -llibtiff -llibopenjp2 -lIlmImf -lzlib -ltegra_hal -framework OpenCL -lAccelerate.framework -lm -ldl -framework Cocoa -framework AppKit -lIconv::Iconv` are not static libs and therefore not embedded in `libopencv_<module>.a` libraries.

In your `CMakeLists.txt` file I would suggest to use `pkg_check_modules` function (see cmake [FindPkgConfig](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html)) to get the same result as `pkg-config --cflags --libs --static opencv4` with something like 
```
pkg_check_modules(OPENCV REQUIRED opencv4)
```
that should set the following output vars `OPENCV_INCLUDE_DIRS`, `OPENCV_STATIC_LIBRARIES` and `OPENCV_LIBRARIES`.

Som5ra, 2024-09-04:
Thanks a lot! I have tried it but encounter another question:

```
CMake Error at /usr/local/lib/cmake/opencv4/OpenCVModules.cmake:375 (set_target_properties):
  The link interface of target "opencv_wechat_qrcode" contains:

    Iconv::Iconv

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/local/lib/cmake/opencv4/OpenCVConfig.cmake:126 (include)
  cmake/VISPUtils.cmake:541 (find_package)
  CMakeLists.txt:655 (VP_OPTION)
```

Then I rebuild opencv without opencv_wechat_qrcode module, then try to build visp again:

```
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/testXmlParserCamera] Error 1
make[1]: *** [modules/core/CMakeFiles/testXmlParserCamera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 21%] Linking CXX executable testCameraParametersConversion
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/testCameraParametersConversion] Error 1
make[1]: *** [modules/core/CMakeFiles/testCameraParametersConversion.dir/all] Error 2
[ 21%] Linking CXX executable testJsonCamera
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/testJsonCamera] Error 1
make[1]: *** [modules/core/CMakeFiles/testJsonCamera.dir/all] Error 2
[ 21%] Linking CXX executable perfGaussianFilter
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/perfGaussianFilter] Error 1
make[1]: *** [modules/core/CMakeFiles/perfGaussianFilter.dir/all] Error 2
[ 21%] Linking CXX executable perfImageAddSub
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/perfImageAddSub] Error 1
make[1]: *** [modules/core/CMakeFiles/perfImageAddSub.dir/all] Error 2
[ 21%] Linking CXX executable perfColorConversion
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/perfColorConversion] Error 1
make[1]: *** [modules/core/CMakeFiles/perfColorConversion.dir/all] Error 2
[ 21%] Linking CXX executable perfImageResize
[ 21%] Linking CXX executable perfImageMorphology
ld: library not found for -lcrt0.o
ld: library not found for -lcrt0.o
clang: clang: error: error: linker command failed with exit code 1 (use -v to see invocation)
linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [modules/core/perfImageMorphology] Error 1
make[2]: *** [modules/core/perfImageResize] Error 1
make[1]: *** [modules/core/CMakeFiles/perfImageMorphology.dir/all] Error 2
make[1]: *** [modules/core/CMakeFiles/perfImageResize.dir/all] Error 2
make: *** [all] Error 2
```

I searched it and found this: https://forum.crystal-lang.org/t/cannot-compile-statically-ld-library-not-found-for-lcrt0-o/4070/2, have you ever met this?

My modification:
In VISPUtils.cmake, add this in https://github.com/lagadic/visp/blob/master/cmake/VISPUtils.cmake#L502:
```
  if("${__first_package}" STREQUAL "OpenCV")
    # set(OpenCV_STATIC ON)
    # set(CMAKE_EXE_LINKER_FLAGS "-static")
    pkg_check_modules(OPENCV REQUIRED opencv4)
  endif()
```

modify https://github.com/lagadic/visp/blob/master/cmake/templates/VISPConfig.cmake.in#L377 to:
```
target_link_libraries(${target} ${VISP_LIBRARIES} ${OPENCV_STATIC_LIBRARIES})
```

Thanks again!

Som5ra, 2024-09-04:
Hi @fspindle , I some more updates about this issue.

I just solved the 'lcrt0.o' problem by disable building all the executable program (with main() entry) like example, test, tutorial. Only keep building the visp components and my own .bundle file. It contains opencv component + visp component + my own code and works well now when called by C# side.

But because of this 'lcrt0.o', I cannot static compile any executable file, which means I cannot debug. Is there any solution about this? 

https://forum.crystal-lang.org/t/cannot-compile-statically-ld-library-not-found-for-lcrt0-o/4070/2, people here say MACOS does not support pure static build, but I don't get why I can static build visp base on a dynamic opencv, then ofc I would get a .bundle including visp components by doing so but need prepare opencv additionally.

Thanks a lot!

fspindle, 2024-09-05:
No idea concerning `lcrt0.o`.
For debugging, you can try on Ubuntu...

Som5ra, 2024-09-06:
> No idea concerning `lcrt0.o`. For debugging, you can try on Ubuntu...

Really thanks, I successfully built the static .so and the .bundle.

I am still wondering that what extra steps I need to do if I want use the .so/.bundle to write C# in unity, then build the whole unity project into ios/android side. For example in ios, do I need to build a specific visp version for ios? and opencv for ios? or I need to disable some modules from visp or opencv? Do you have any advice for this? 

For now the phone side just connot load the dll after I built the ios project, it says
```
DllNotFoundException: Unable to load DLL 'UnityWrapper'. Tried the load the following dynamic libraries: Unable to load dynamic library '/UnityWrapper' because of 'Failed to open the requested dynamic library (0x06000000) dlerror() = dlopen(/UnityWrapper, 0x0005): tried: '/UnityWrapper' (no such file), '/private/preboot/Cryptexes/OS/UnityWrapper' (no such file), '/UnityWrapper' (no such file)
```

Thanks a lot in advance!!

elvis75k, 2024-10-10:
> > No idea concerning `lcrt0.o`. For debugging, you can try on Ubuntu...
> 
> Really thanks, I successfully built the static .so and the .bundle.
> 
> I am still wondering that what extra steps I need to do if I want use the static .so/.bundle to write C# in unity, then build the whole unity project into ios/android side. For example in ios, do I need to build a specific visp version for ios? and opencv for ios? or I need to disable some modules from visp or opencv? Do you have any advice for this?
> 
> For now the phone side just connot load the dll after I built the ios project, it says
> 
> DllNotFoundException: Unable to load DLL 'UnityWrapper'. Tried the load the following dynamic libraries: Unable to load dynamic library '/UnityWrapper' because of 'Failed to open the requested dynamic library (0x06000000) dlerror() = dlopen(/UnityWrapper, 0x0005): tried: '/UnityWrapper' (no such file), '/private/preboot/Cryptexes/OS/UnityWrapper' (no such file), '/UnityWrapper' (no such file)
> 
> Thanks a lot in advance!!

While testing ViSP for the first time, i'm having the exact same error in the XCode's. This is happening when testing the Unity Visp wrapper tutorial project build for ios platform. In editor works just fine though.

What it missing to generate an app that works on ios from unity?
Any hint for building from Unity on iOS/Android would be very helpful.

karthiTD, 2025-02-03:
Does visp unity wrapper plugin (UnityVisp.dll) support mobile platforms (Android/iOS) or only for windows desktop app?

Som5ra, 2025-02-04:
@elvis75k @karthiTD 

I think some modules do support mobile devices, you can try building visp module by module to test, for instance, visp_core, you can write a debug function in the c++ source code then use it in unity.

For those modules that may involve 3rd party libraries, it depends on whether the specific 3rd party lib support mobile arch or not. If it does like opencv, you should build a static version of it, then try linking it when build visp.

Even so, I can not make sure it can be built and run correctly. Sry for the late reply.

closed 2025-02-04 by Som5ra