lagadic/visp #1477: Can't find visp_mbt361.dll, visp_io361.dll, visp_core361.dll, visp_gui361.dll on Windows10 based on c++ install.

LeonFight, 2024-10-04:
I have installed VISP from source for Windows 10 with Visual C++ 2022 (vc17), following all the steps of the website and try to get started. But when I tried to run tutorial-viewer example
`%VISP_WS%\image\build\> tutorial-viewer.exe monkey.pgm`
It cames errors of can't find visp_mbt361.dll, visp_io361.dll, visp_core361.dll, visp_gui361.dll.

The sames errors also appeared when I tried "Advanced getting started".

[image]

fspindle, 2024-10-04:
This is because you didn't set `PATH` env var with the location of visp DLLs as explained in the [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc17.html#install_win10_msvc17_env_var).

fspindle, 2024-10-04:
`VISP_DIR` is the env var to help cmake to find ViSP.
You have also to set `PATH` env var adding the path to ViSP DLLs folder location. In you case you should run something like:
```
setx PATH %PATH%;"E:\gkl\visp-ws\visp-build-vc17\x64\vc17\bin
```
Now the following should work
```
tutorial-viewer ..\..\monkey.ppm 
```
If this is not the case, open a fresh `cmd` terminal and try again

LeonFight, 2024-10-04:
I ran:

`setx VISP_WINDOWS_DLL_PATH "%VISP_WS%\visp-build-vc17\x64\vc17\bin`

Then close the `cmd` and open a fresh `cmd` ran `tutorial-viewer ..\..\monkey.ppm`. However it does not work.

I agree that this path sould be added to ViSP path env, but wondering what's the exact path name of it.

LeonFight, 2024-10-06:
This issue is solved. I changed the env name `Path` to `PATH` then it works.

closed 2024-10-06 by LeonFight