lagadic/visp #1487: Segfault with vpCannyEdgeDetection with a certain image

s-trinh, 2024-10-21:
Hello,

With the following image, I encounter a segfault with ViSP [`v3.6.0-1651-ga65eebc2f-dirty`](https://github.com/lagadic/visp/tree/a65eebc2fffc3814889fbe40c3b1a8ee182d1bd4).
[image]

Code to reproduce the issue:
```c++
#include <iostream>
#include <visp3/core/vpConfig.h>
#include <visp3/core/vpIoTools.h>
#include <visp3/core/vpCannyEdgeDetection.h>
#include <visp3/io/vpImageIo.h>

int main(int argc, const char **argv)
{
  std::cout << "getBuildInformation()\n" << vpIoTools::getBuildInformation() << std::endl;

  int gaussianKernelSize = 3;
  float gaussianStdev = 0.5f;
  int apertureSize = 3;
  // Canny parameters
  float lowerThresh = -1.;
  float upperThresh = -1.;
  float lowerThreshRatio = 0.6f;
  float upperThreshRatio = 1.5f;
  vpImageFilter::vpCannyFilteringAndGradientType filteringType = vpImageFilter::CANNY_GBLUR_SOBEL_FILTERING;

  vpCannyEdgeDetection cannyDetector(gaussianKernelSize, gaussianStdev, apertureSize,
                                    lowerThresh, upperThresh, lowerThreshRatio, upperThreshRatio,
                                    filteringType);

  vpImage<unsigned char> I_gray;
  vpImageIo::read(I_gray, "/tmp/img.png");
  std::cout << "I_gray=" << I_gray.getWidth() << "x" << I_gray.getHeight() << std::endl;

  vpImage<unsigned char> I_canny_visp;
  I_canny_visp = cannyDetector.detect(I_gray);

  return EXIT_SUCCESS;
}
```

gdb stack trace:
```bash
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff77fe8c1 in __dcigettext (domainname=0x0, msgid1=0x7ffff7ecec00 "map::at", msgid2=0x0, plural=0, n=0, category=5) at dcigettext.c:521
521	dcigettext.c: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  0x00007ffff77fe8c1 in __dcigettext (domainname=0x0, msgid1=0x7ffff7ecec00 "map::at", msgid2=0x0, plural=0, n=0, category=5) at dcigettext.c:521
#1  0x00007ffff7a7731b in std::__throw_out_of_range(char const*) () at /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007ffff7ca5688 in std::map<std::pair<unsigned int, unsigned int>, vpCannyEdgeDetection::EdgeType, std::less<std::pair<unsigned int, unsigned int> >, std::allocator<std::pair<std::pair<unsigned int, unsigned int> const, vpCannyEdgeDetection::EdgeType> > >::at(std::pair<unsigned int, unsigned int> const&) (__k={...}, this=0x7fffffffdca8) at /usr/include/c++/9/bits/stl_pair.h:454
#3  vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:547
#4  0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#5  0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#6  0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#7  0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#8  0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#9  0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#10 0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#11 0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#12 0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
    at visp/modules/core/src/image/vpCannyEdgeDetection.cpp:554
#13 0x00007ffff7ca57a6 in vpCannyEdgeDetection::recursiveSearchForStrongEdge(std::pair<unsigned int, unsigned int> const&) (this=0x7fffffffdb60, coordinates={...})
```

Looks like using the following guard does not help:
```bash
diff --git a/modules/core/src/image/vpCannyEdgeDetection.cpp b/modules/core/src/image/vpCannyEdgeDetection.cpp
index 310814709..ac727c9ce 100644
--- a/modules/core/src/image/vpCannyEdgeDetection.cpp
+++ b/modules/core/src/image/vpCannyEdgeDetection.cpp
@@ -544,14 +544,16 @@ vpCannyEdgeDetection::recursiveSearchForStrongEdge(const std::pair<unsigned int,
         try {
           std::pair<unsigned int, unsigned int> key_candidate(idRow, idCol);
           // Checking if the 8-neighbor point is in the list of edge candidates
-          EdgeType type_candidate = m_edgePointsCandidates.at(key_candidate);
-          if (type_candidate == STRONG_EDGE) {
-            // The 8-neighbor point is a strong edge => the weak edge becomes a strong edge
-            hasFoundStrongEdge = true;
-          }
-          else if (type_candidate == WEAK_EDGE) {
-            // Checking if the WEAK_EDGE neighbor has a STRONG_EDGE neighbor
-            hasFoundStrongEdge = recursiveSearchForStrongEdge(key_candidate);
+          if (m_edgePointsCandidates.find(key_candidate) != m_edgePointsCandidates.end()) {
+            EdgeType type_candidate = m_edgePointsCandidates.at(key_candidate);
+            if (type_candidate == STRONG_EDGE) {
+              // The 8-neighbor point is a strong edge => the weak edge becomes a strong edge
+              hasFoundStrongEdge = true;
+            }
+            else if (type_candidate == WEAK_EDGE) {
+              // Checking if the WEAK_EDGE neighbor has a STRONG_EDGE neighbor
+              hasFoundStrongEdge = recursiveSearchForStrongEdge(key_candidate);
+            }
           }
         }
         catch (...) {
```

rolalaro, 2024-10-28:
Hello,
That's weird I cannot reproduce your error using the same parameters.
Could you please run the following:
```
$ cd $VISP_WS/visp-build/tutorial/image
$ ./tutorial-canny -i /tmp/img.png --ratio 0.6 1.5 --gradient 3 0.5 --aperture 3
```
On my computer (Ubuntu 22.04) the program works correctly (even if the Canny results are a bit odd due to the exposure of the image)

s-trinh, 2024-10-28:
Yes I have the same issue with `./tutorial-canny -i /tmp/img.png --ratio 0.6 1.5 --gradient 3 0.5 --aperture 3`

<details>
<summary>ViSP-third-party.txt</summary>

```bash

==========================================================
General configuration information for ViSP 3.6.1

  Version control:               v3.6.0-1651-ga65eebc2f

  Platform:
    Timestamp:                   2023-11-15T22:24:52Z
    Host:                        Linux 5.15.0-84-generic x86_64
    CMake:                       3.16.3
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RelWithDebInfo

  System information:
    Number of CPU logical cores: 16
    Number of CPU physical cores: 8
    Total physical memory (in MiB): 15689
    OS name:                     Linux
    OS release:                  5.15.0-124-generic
    OS version:                  #134~20.04.1-Ubuntu SMP Tue Oct 1 15:27:33 UTC 2024
    OS platform:                 x86_64
    CPU name:                    Unknown P6 family
    Is the CPU 64-bit?           yes
    Does the CPU have FPU?       yes
    CPU optimization:            SSE2 SSE3 SSSE3

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                /usr/bin/c++  (ver 9.4.0)
    C++ flags (Release):         -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibility=hidden -mavx -fPIC -O3 -DNDEBUG
    C++ flags (Debug):           -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibility=hidden -mavx -fPIC -g
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibility=hidden -mavx -fPIC -O3 -DNDEBUG
    C flags (Debug):             -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibility=hidden -mavx -fPIC -g
    Linker flags (Release):
    Linker flags (Debug):
    Use cxx standard:            17

  ViSP modules:
    To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 java
    Enable visp namespace:       no
    Enable explicit keyword:     no

  Python 3:
    Interpreter:                 /usr/bin/python3.8 (ver 3.8.10)

  Java:                          
    ant:                         no
    JNI:                         no

  Python3 bindings:              no
    Requirements: 
      Python version > 3.7.0:    ok (ver 3.8.10)
      Python in Virtual environment or conda:
                                 failed
      Pybind11 found:            failed
      CMake > 3.19.0:            failed (3.16.3)
      C++ standard > 201703L:    ok (201703L)

  Build options: 
[... 926 lines cut ...]
==48425==    by 0x8E11D12: g_param_type_register_static (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==48425==    by 0x8E147EA: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==48425==    by 0x8DFF0CB: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==48425==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==48425==    by 0x4011CA0: call_init (dl-init.c:30)
==48425==    by 0x4011CA0: _dl_init (dl-init.c:119)
==48425==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==48425==    by 0xA: ???
==48425==    by 0x1FFF000156: ???
==48425== 
==48425== LEAK SUMMARY:
==48425==    definitely lost: 0 bytes in 0 blocks
==48425==    indirectly lost: 0 bytes in 0 blocks
==48425==      possibly lost: 1,352 bytes in 18 blocks
==48425==    still reachable: 112,926,670 bytes in 1,285,632 blocks
==48425==                       of which reachable via heuristic:
==48425==                         newarray           : 1,536 bytes in 16 blocks
==48425==         suppressed: 0 bytes in 0 blocks
==48425== Reachable blocks (those to which a pointer was found) are not shown.
==48425== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==48425== 
==48425== ERROR SUMMARY: 18 errors from 18 contexts (suppressed: 0 from 0)
```

</details>

rolalaro, 2024-10-29:
I suspect that the high threshold is too high, leading to all the edges being considered as weak edges. Due to that, the `vpCannyEdgeDetection::recursiveSearchForStrongEdge` could be cold until the stack of calls is too high to be handled. Could you please replace the code of this method by the following, compile and give me the result of the print please ? Basically, the instead of calling  `vpCannyEdgeDetection::recursiveSearchForStrongEdge`, the method `vpCannyEdgeDetection::performEdgeTracking` will count the number of weak and strong edges.

```
void
vpCannyEdgeDetection::performEdgeTracking()
{
  std::map<std::pair<unsigned int, unsigned int>, EdgeType>::iterator it;
  std::map<std::pair<unsigned int, unsigned int>, EdgeType>::iterator m_edgePointsCandidates_end = m_edgePointsCandidates.end();
  unsigned int nbStrongEdges = 0, nbWeakEdges = 0;
  for (it = m_edgePointsCandidates.begin(); it != m_edgePointsCandidates_end; ++it) {
    if (it->second == STRONG_EDGE) {
      m_edgeMap[it->first.first][it->first.second] = 255;
      if (m_storeListEdgePoints) {
        m_edgePointsList.push_back(vpImagePoint(it->first.first, it->first.second));
      }
      ++nbStrongEdges;
    }
    else if (it->second == WEAK_EDGE) {
      // if (recursiveSearchForStrongEdge(it->first)) {
      m_edgeMap[it->first.first][it->first.second] = 255;
    // }
      ++nbWeakEdges;
    }
  }
  std::cout << "[vpCannyEdgeDetection::performEdgeTracking]#strong edges = " << nbStrongEdges << std::endl;
  std::cout << "[vpCannyEdgeDetection::performEdgeTracking]#weak edges = " << nbWeakEdges << std::endl;
}
```

s-trinh, 2024-10-29:
The output:

```bash
I_gray=1788x942
[vpCannyEdgeDetection::performEdgeTracking]#strong edges = 102138
[vpCannyEdgeDetection::performEdgeTracking]#weak edges = 539881
```

rolalaro, 2024-10-30:
OK not what I expected to be honnest, I was expected something like 0 strong edges and a lot of weak edges.
What's weird is that I cannot reproduce it, the  Mac OS CIs worked but the Ubuntu 22.04 CIs failed.

The differences between our configurations:
- *Ubuntu versions*: your computer is on 20.04 whereas my computer and the Linux CIs is on Ubuntu 22.04 
- *c++ compiler versions*: you have c++ 9.4 while I have c++ 11.4
- *Blas/Lapack*: you use Lapack built-in while I use Atlas
- *Eigen versions*: you have 3.3.7 while I have 3.4.0

By any chance do you have Docker ? If so, could you create an image based on Ubuntu 22.04 with the same c++ compiler version than I have to see if it changes anything ? It might not, because it seems that the OS release is different between the Docker images and the one that is installed on my computer.

Other question: have you ever faced this segfault with other images ? Such as a with image with a gray empty square inside ?

rolalaro, 2024-10-30:
After reflexion, I will create a Docker that reproduces your configuration, try to reproduce the bug and come back to you if that's fine with you ?

rolalaro, 2024-10-30:
OK I downloaded the image you posted (before I did a screenshot because I didn't find how to save it) and now I have a segfault on my computer too. So it's not due to a different configuration, there is a segfault happening somewhere. I'll continue to dig in.

rolalaro, 2024-10-30:
Found the problem !
My original thought was correct, the upper threshold ratio you gave was wrong. It must be between 0 and 1, because it corresponds to the ratio of points that must have the norm of their gradient lower than the found threshold. In other words, if you set the upper ratio to 0.5, it means that at least 50% of the pixels will have a gradient lower than the found threshold.

See   #1490 for the fix.

s-trinh, 2024-10-30:
Thanks for your quick feedback.
I also have to refresh the webpage sometimes in order to open the image in another tab on Github.

---

It looks like for certain values of Gaussian std and for some over-exposed images we are hitting somehow the limit of the stack size.

For instance when using this image or the image from the first post

[image]

- `./tutorial-canny -i /tmp/2015_03449.png --gradient 3 0.5`:

```bash
Canny Configuration:
	Filtering + gradient operators = gaussianblur+sobel-filtering
	Gaussian filter kernel size = 3
	Gaussian filter standard deviation = 0.500000
	Gradient filter kernel size = 3
	Canny edge filter thresholds = [-1.000000 ; -1.000000]
	Canny edge filter thresholds ratio (for auto-thresholding) = [0.600000 ; 0.800000]
```

- dIx_uchar
[image]

- dIy_uchar
[image]

- without the `recursiveSearchForStrongEdge()` it gives:

```bash
[vpCannyEdgeDetection::performEdgeTracking]#strong edges = 77555
[vpCannyEdgeDetection::performEdgeTracking]#weak edges = 264432
```

- I_canny_visp (the top and bottom white bands are I guess what makes the call to `recursiveSearchForStrongEdge()` hitting the stack size limit)

[image]

It works correctly with `./tutorial-canny -i /tmp/2015_03449.png --gradient 3 0.1`.

---

Increasing the stack size make it works.

```bash
ulimit -a
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 62389
max locked memory       (kbytes, -l) 65536
max memory size         (kbytes, -m) unlimited
open files                      (-n) 1024
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 0
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 62389
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited

ulimit -s 16384

ulimit -a
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 62389
max locked memory       (kbytes, -l) 65536
max memory size         (kbytes, -m) unlimited
open files                      (-n) 1024
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 0
stack size              (kbytes, -s) 16384
cpu time               (seconds, -t) unlimited
max user processes              (-u) 62389
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited

./tutorial-canny -i /tmp/2015_03449.png --gradient 3 0.5
Canny Configuration:
	Filtering + gradient operators = gaussianblur+sobel-filtering
	Gaussian filter kernel size = 3
	Gaussian filter standard deviation = 0.500000
	Gradient filter kernel size = 3
	Canny edge filter thresholds = [-1.000000 ; -1.000000]
	Canny edge filter thresholds ratio (for auto-thresholding) = [0.600000 ; 0.800000]
```

And the result image:

[image]

fspindle, 2024-10-30:
This issue seems definitively due to the recursive algorithm. I remember that in vpDot that implements also a recursive algorithm, we had also issues with the size of the stack.

The fix that I found on that time was to add in vpDot.cpp. 

```
#ifdef VISP_USE_MSVC
#pragma comment(linker, "/STACK:256000000") // Increase max recursion depth
#endif
```

Similar code could be tested to see if it helps.

rolalaro, 2024-10-31:
@s-trinh oh indeed the results largely differ depending on the gradient standard deviation ! My bad, I tested the algorithm on some images coming from a dataset known for testing Canny implementations, but I didn't think to try it on overexposed images.
I will try the solution proposed by @fspindle and let you know the results.

rolalaro, 2024-11-05:
@s-trinh the following proposition has been retained and is implemented in  #1490 :
- for Windows, the stack size is increased at compilation time, because as far as I know it cannot be increase at runtime 
- for POSIX systems, the user can change the required stack size using the method `setMinimumStackSize` . At runtime, we check if the stack size is at least as big as the minimum required stack size. If not, the stack size is increased until the end of the call to the `detect` method.

Can you tell us if this solution answers your needs, to know if we can close this issue ?

s-trinh, 2024-11-05:
@rolalaro 

Yes, the issue is solved now.

fspindle, 2024-11-06:
Fixed in #1490

closed 2024-11-06 by fspindle