lagadic/visp #1552: Unabled to use SIFT feature with opencv 4.5.4

cleprt, 2025-01-21:
I am using visp on Ubuntu 22 with opencv 4.5.4 (installed via apt libopencv-dev). 

The use of vpKeyPoint::initDetector(const std::string &detectorName) with "SIFT" fails in the new master branch while it worked for me on the commit 70ddb40b68c1d61a2544e36eb113a9f3a758d8b7 with the same opencv version.

I feel like the line https://github.com/lagadic/visp/blob/master/modules/vision/src/key-point/vpKeyPoint.cpp#L1784 does not take all cases in consideration as with opencv 4.5.4 I do have the file opencv2/features2d/features2d.hpp (which means def of HAVE_OPENCV_FEATURES) and not xfeatures2d.hpp.

fspindle, 2025-01-22:
This issue is to make in relation to [PR #1522](https://github.com/lagadic/visp/pull/1522) that introduced compat with OpenCV 5.x.

fspindle, 2025-01-22:
@cleprt Should be fixed in PR #1555

commit 8b7a7e4c (2025-01-23): Fix macro used to enable/disable SIFT usage depending on OpenCV version. Closes #1552
closed 2025-01-23 by fspindle