lagadic/visp #1594: Can you provide a dockerfile that contains the Python interface for compilation?

GavinYang5, 2025-02-24:
I tried to follow the tutorial to build a Python wrapper, but I failed
```
/home/vispci/miniforge3/envs/visp-conda-ws/include/X11/Xlib.h:44:10:  fatal error: X11/X.h: No such file or directory
44 | #include <X11/X.h>
|          ^~~~~~~~~
compilation terminated.
```
Can it be built directly in Dockerfile

What should I do?

fspindle, 2025-02-24:
Can you give us more details to reproduce your issue...

olivier-roussel, 2025-02-24:
Could you send us the output of both `conda info` and `conda list` commands when your `visp-conda-ws` conda environment is activated ?

GavinYang5, 2025-02-25:
`-- Detected processor: x86_64
-- pybind11 found
-- Could NOT find MKL (missing: MKL_LIBRARIES MKL_INCLUDE_DIRS MKL_INTERFACE_LIBRARY MKL_SEQUENTIAL_LAYER_LIBRARY MKL_CORE_LIBRARY) 
-- Could not find OpenBLAS include. Turning OpenBLAS_FOUND off
-- Could not find OpenBLAS lib. Turning OpenBLAS_FOUND off
-- Could NOT find Atlas (missing: Atlas_CBLAS_INCLUDE_DIR Atlas_CLAPACK_INCLUDE_DIR Atlas_BLAS_LIBRARY) 
Package lapack was not found in the pkg-config search path.
Perhaps you should add the directory containing `lapack.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lapack' found
-- Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY) 
-- Try C99 flag = [-std=gnu99]
-- Performing Test C99_FLAG_DETECTED
-- Performing Test C99_FLAG_DETECTED - Success
-- dc1394 found
-- Could NOT find FTIITSDK (missing: FTIITSDK_FT_LIBRARY FTIITSDK_COMMUNICATION_LIBRARY FTIITSDK_INCLUDE_DIR) 
-- openmp found
-- eigen3 found
-- Could NOT find Panda3D (missing: ALL_LIBS_FOUND Panda3D_INCLUDE_DIRS) 
-- Checking for module 'OGRE'
--   No package 'OGRE' found
-- realsense2 found
-- threads found
-- xml2 found
-- opencv found
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- x11 found
-- jpeg found
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- Could NOT find PNG (missing: PNG_PNG_INCLUDE_DIR) 
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- png found
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS) 
-- nlohmann_json found
-- Performing Test HAVE_FUNC_ISNAN
-- Performing Test HAVE_FUNC_ISNAN - Success
-- Performing Test HAVE_FUNC_STD_ISNAN
-- Performing Test HAVE_FUNC_STD_ISNAN - Success
-- Performing Test HAVE_FUNC__ISNAN
-- Performing Test HAVE_FUNC__ISNAN - Failed
-- Performing Test HAVE_FUNC_ISINF
-- Performing Test HAVE_FUNC_ISINF - Success
-- Performing Test HAVE_FUNC_STD_ISINF
-- Performing Test HAVE_FUNC_STD_ISINF - Success
-- Performing Test HAVE_FUNC_ROUND
-- Performing Test HAVE_FUNC_ROUND - Success
-- Performing Test HAVE_FUNC_STD_ROUND
-- Performing Test HAVE_FUNC_STD_ROUND - Success
-- Performing Test HAVE_FUNC_ERFC
-- Performing Test HAVE_FUNC_ERFC - Success
-- Performing Test HAVE_FUNC_STD_ERFC
-- Performing Test HAVE_FUNC_STD_ERFC - Success
-- Performing Test HAVE_FUNC_ISFINITE
-- Performing Test HAVE_FUNC_ISFINITE - Success
-- Performing Test HAVE_FUNC_STD_ISFINITE
-- Performing Test HAVE_FUNC_STD_ISFINITE - Success
-- Performing Test HAVE_FUNC__FINITE
-- Performing Test HAVE_FUNC__FINITE - Failed
-- Performing Test HAVE_FUNC_LOG1P
-- Performing Test HAVE_FUNC_LOG1P - Success
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Could NOT find JNI (missing: JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 AWT JVM) 
-- 
-- ==========================================================
-- General configuration information for ViSP 3.6.1
-- 
--   Version control:               v3.6.0-2182-ge8195fbad
-- 
--   Platform:
[... 180 lines cut ...]
--     Use TensorRT:                no
-- 
--   Documentation: 
--     Use doxygen:                 no
--     \- Use mathjax:              no
-- 
--   Tests and samples:
--     Use catch2 (built-in):       yes (ver 3.7.1)
--     Tests:                       yes
--     Apps:                        yes
--     Demos:                       yes
--     Examples:                    yes
--     Tutorials:                   yes
--     Dataset found:               yes (ver 3.7.0 in /home/vispci/visp-ws/visp-images)
-- 
--   Library dirs:
--     Eigen3 dir:                  /home/vispci/miniforge3/envs/visp-conda-ws/share/eigen3/cmake
--     OpenCV dir:                  /home/vispci/miniforge3/envs/visp-conda-ws/lib/cmake/opencv4
-- 
--   Install path:                  /home/vispci/miniforge3/envs/visp-conda-ws
-- 
-- ==========================================================
-- Configuring done (2.8s)
-- Generating done (2.0s)
-- Build files have been written to: /home/vispci/visp-ws/visp-build-bindings`

GavinYang5, 2025-02-25:
-- Detected processor: x86_64
-- pybind11 found
-- Could NOT find MKL (missing: MKL_LIBRARIES MKL_INCLUDE_DIRS MKL_INTERFACE_LIBRARY MKL_SEQUENTIAL_LAYER_LIBRARY MKL_CORE_LIBRARY) 
-- Could not find OpenBLAS include. Turning OpenBLAS_FOUND off
-- Could not find OpenBLAS lib. Turning OpenBLAS_FOUND off
-- Could NOT find Atlas (missing: Atlas_CBLAS_INCLUDE_DIR Atlas_CLAPACK_INCLUDE_DIR Atlas_BLAS_LIBRARY) 
Package lapack was not found in the pkg-config search path.
Perhaps you should add the directory containing `lapack.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lapack' found
-- Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY) 
-- Try C99 flag = [-std=gnu99]
-- Performing Test C99_FLAG_DETECTED
-- Performing Test C99_FLAG_DETECTED - Success
-- dc1394 found
-- Could NOT find FTIITSDK (missing: FTIITSDK_FT_LIBRARY FTIITSDK_COMMUNICATION_LIBRARY FTIITSDK_INCLUDE_DIR) 
-- openmp found
-- eigen3 found
-- Could NOT find Panda3D (missing: ALL_LIBS_FOUND Panda3D_INCLUDE_DIRS) 
-- Checking for module 'OGRE'
--   No package 'OGRE' found
-- realsense2 found
-- threads found
-- xml2 found
-- opencv found
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- x11 found
-- jpeg found
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- Could NOT find PNG (missing: PNG_PNG_INCLUDE_DIR) 
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- png found
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS) 
-- nlohmann_json found
-- Performing Test HAVE_FUNC_ISNAN
-- Performing Test HAVE_FUNC_ISNAN - Success
-- Performing Test HAVE_FUNC_STD_ISNAN
-- Performing Test HAVE_FUNC_STD_ISNAN - Success
-- Performing Test HAVE_FUNC__ISNAN
-- Performing Test HAVE_FUNC__ISNAN - Failed
-- Performing Test HAVE_FUNC_ISINF
-- Performing Test HAVE_FUNC_ISINF - Success
-- Performing Test HAVE_FUNC_STD_ISINF
-- Performing Test HAVE_FUNC_STD_ISINF - Success
-- Performing Test HAVE_FUNC_ROUND
-- Performing Test HAVE_FUNC_ROUND - Success
-- Performing Test HAVE_FUNC_STD_ROUND
-- Performing Test HAVE_FUNC_STD_ROUND - Success
-- Performing Test HAVE_FUNC_ERFC
-- Performing Test HAVE_FUNC_ERFC - Success
-- Performing Test HAVE_FUNC_STD_ERFC
-- Performing Test HAVE_FUNC_STD_ERFC - Success
-- Performing Test HAVE_FUNC_ISFINITE
-- Performing Test HAVE_FUNC_ISFINITE - Success
-- Performing Test HAVE_FUNC_STD_ISFINITE
-- Performing Test HAVE_FUNC_STD_ISFINITE - Success
-- Performing Test HAVE_FUNC__FINITE
-- Performing Test HAVE_FUNC__FINITE - Failed
-- Performing Test HAVE_FUNC_LOG1P
-- Performing Test HAVE_FUNC_LOG1P - Success
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Could NOT find JNI (missing: JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 AWT JVM) 
-- 
-- ==========================================================
-- General configuration information for ViSP 3.6.1
-- 
--   Version control:               v3.6.0-2182-ge8195fbad
-- 
--   Platform:
[... 180 lines cut ...]
--     Use TensorRT:                no
-- 
--   Documentation: 
--     Use doxygen:                 no
--     \- Use mathjax:              no
-- 
--   Tests and samples:
--     Use catch2 (built-in):       yes (ver 3.7.1)
--     Tests:                       yes
--     Apps:                        yes
--     Demos:                       yes
--     Examples:                    yes
--     Tutorials:                   yes
--     Dataset found:               yes (ver 3.7.0 in /home/vispci/visp-ws/visp-images)
-- 
--   Library dirs:
--     Eigen3 dir:                  /home/vispci/miniforge3/envs/visp-conda-ws/share/eigen3/cmake
--     OpenCV dir:                  /home/vispci/miniforge3/envs/visp-conda-ws/lib/cmake/opencv4
-- 
--   Install path:                  /home/vispci/miniforge3/envs/visp-conda-ws
-- 
-- ==========================================================
-- Configuring done (2.8s)
-- Generating done (2.0s)
-- Build files have been written to: /home/vispci/visp-ws/visp-build-bindings

GavinYang5, 2025-02-25:
(visp-conda-ws) vispci@:~/visp-ws/visp-build-bindings$ conda info

     active environment : visp-conda-ws
    active env location : /home/vispci/miniforge3/envs/visp-conda-ws
            shell level : 6
       user config file : /home/vispci/.condarc
 populated config files : /home/vispci/miniforge3/.condarc
          conda version : 24.11.3
    conda-build version : not installed
         python version : 3.12.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=icelake
                          __conda=24.11.3=0
                          __cuda=11.4=0
                          __glibc=2.31=0
                          __linux=5.15.0=0
                          __unix=0=0
       base environment : /home/vispci/miniforge3  (writable)
      conda av data dir : /home/vispci/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/vispci/miniforge3/pkgs
                          /home/vispci/.conda/pkgs
       envs directories : /home/vispci/miniforge3/envs
                          /home/vispci/.conda/envs
               platform : linux-64
             user-agent : conda/24.11.3 requests/2.32.3 CPython/3.12.8 Linux/5.15.0-131-generic ubuntu/20.04.6 glibc/2.31 solver/libmamba conda-libmamba-solver/24.9.0 libmambapy/1.5.12
                UID:GID : 1001:1001
             netrc file : None
           offline mode : False

GavinYang5, 2025-02-25:
# packages in environment at /home/vispci/miniforge3/envs/visp-conda-ws:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
alsa-lib                  1.2.13               hb9d3cd8_0    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
binutils                  2.43                 h4852527_4    conda-forge
binutils_impl_linux-64    2.43                 h4bf12b8_4    conda-forge
binutils_linux-64         2.43                 h4852527_4    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.34.4               hb9d3cd8_0    conda-forge
c-compiler                1.9.0                h2b85faf_0    conda-forge
ca-certificates           2025.1.31            hbcca054_0    conda-forge
cairo                     1.18.2               h3394656_1    conda-forge
cmake                     3.31.5               h74e3db0_0    conda-forge
cudatoolkit               11.8.0              h4ba93d1_13    conda-forge
cxx-compiler              1.9.0                h1a2810e_0    conda-forge
cyrus-sasl                2.1.27               h54b06d7_7    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
double-conversion         3.3.1                h5888daf_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
expat                     2.6.4                h5888daf_0    conda-forge
ffmpeg                    7.1.0           gpl_hbc7a4c3_113    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h7e30c49_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                ha6d2627_3    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gcc                       13.3.0               h9576a4e_2    conda-forge
gcc_impl_linux-64         13.3.0               h1e990d8_2    conda-forge
gcc_linux-64              13.3.0               hc28eda2_7    conda-forge
gdk-pixbuf                2.42.12              hb9ae30d_0    conda-forge
gettext                   0.23.1               h5888daf_0    conda-forge
gettext-tools             0.23.1               h5888daf_0    conda-forge
gmp                       6.3.0                hac33072_2    conda-forge
graphite2                 1.3.13            h59595ed_1003    conda-forge
gxx                       13.3.0               h9576a4e_2    conda-forge
gxx_impl_linux-64         13.3.0               hae580e1_2    conda-forge
gxx_linux-64              13.3.0               h6834431_7    conda-forge
harfbuzz                  10.3.0               h76408a6_0    conda-forge
hdf5                      1.14.3          nompi_h2d575fe_109    conda-forge
icu                       75.1                 he02047a_0    conda-forge
imath                     3.1.12               h7955e40_0    conda-forge
jack                      1.9.22               h7c63dc7_2    conda-forge
jasper                    4.2.4                h536e39c_0    conda-forge
kernel-headers_linux-64   3.10.0              he073ed8_18    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.21.3               h659f571_0    conda-forge
lame                      3.100             h166bdaf_1003    conda-forge
ld_impl_linux-64          2.43                 h712a8e2_4    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
level-zero                1.20.6               h84d6215_0    conda-forge
libabseil                 20240722.0      cxx17_hbbce691_4    conda-forge
libaec                    1.1.3                h59595ed_0    conda-forge
libasprintf               0.23.1               h8e693c7_0    conda-forge
libasprintf-devel         0.23.1               h8e693c7_0    conda-forge
libass                    0.17.3               hba53ac1_1    conda-forge
libavif16                 1.1.1                h1909e37_2    conda-forge
libblas                   3.9.0           31_h59b9bed_openblas    conda-forge
libcap                    2.71                 h39aace5_0    conda-forge
libcblas                  3.9.0           31_he106b2a_openblas    conda-forge
libclang-cpp19.1          19.1.7          default_hb5137d0_1    conda-forge
[... 129 lines cut ...]
x265                      3.5                  h924138e_3    conda-forge
xcb-util                  0.4.1                hb711507_2    conda-forge
xcb-util-cursor           0.1.5                hb9d3cd8_0    conda-forge
xcb-util-image            0.4.0                hb711507_2    conda-forge
xcb-util-keysyms          0.4.1                hb711507_0    conda-forge
[... cut ...]
xorg-libxi                1.8.2                hb9d3cd8_0    conda-forge
xorg-libxrandr            1.5.4                hb9d3cd8_0    conda-forge
xorg-libxrender           0.9.12               hb9d3cd8_0    conda-forge
xorg-libxscrnsaver        1.2.4                hb9d3cd8_0    conda-forge
xorg-libxtst              1.2.5                hb9d3cd8_3    conda-forge
xorg-libxxf86vm           1.1.6                hb9d3cd8_0    conda-forge
zstd                      1.5.7                hb8e6e7a_1    conda-forge

GavinYang5, 2025-02-25:
The above is more detailed information, please help to take a look, thank you

GavinYang5, 2025-02-25:
I solved the problem with X11, but I had a new problem:

Collecting pybind11-stubgen
  Downloading pybind11_stubgen-2.5.3-py3-none-any.whl.metadata (1.9 kB)
Downloading pybind11_stubgen-2.5.3-py3-none-any.whl (29 kB)
Installing collected packages: pybind11-stubgen
Successfully installed pybind11-stubgen-2.5.3
Traceback (most recent call last):
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/site-packages/visp/__init__.py", line 44, in <module>
    from ._visp import *
ImportError: /home/vispci/visp-ws/visp-build-bindings/lib/libvisp_core.so.3.6: undefined symbol: _ZN8vpMatrixaSEd

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/site-packages/pybind11_stubgen/__main__.py", line 4, in <module>
    main()
    ~~~~^^
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/site-packages/pybind11_stubgen/__init__.py", line 319, in main
    run(
    ~~~^
        parser,
        ^^^^^^^
    ...<5 lines>...
        writer=Writer(stub_ext=args.stub_extension),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/site-packages/pybind11_stubgen/__init__.py", line 358, in run
    QualifiedName.from_str(module_name), importlib.import_module(module_name)
                                         ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1310, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/site-packages/visp/__init__.py", line 57, in <module>
    raise ImportError('Could not import ViSP python bindings')
ImportError: Could not import ViSP python bindings
Traceback (most recent call last):
  File "/home/vispci/visp-ws/visp/modules/python/stubs/run_stub_generator.py", line 55, in <module>
    subprocess.run([sys.executable, '-m', 'pybind11_stubgen',  '-o', str(output_root.absolute()), '--ignore-all-errors', 'visp._visp'], check=True)
    ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13/subprocess.py", line 579, in run
    raise CalledProcessError(retcode, process.args,
                             output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/vispci/miniforge3/envs/visp-conda-ws/bin/python3.13', '-m', 'pybind11_stubgen', '-o', '/home/vispci/visp-ws/visp-build-bindings/modules/python/stubs', '--ignore-all-errors', 'visp._visp']' returned non-zero exit status 1.
make[3]: *** [modules/python/stubs/CMakeFiles/visp_python_bindings_stubs.dir/build.make:73: modules/python/stubs/CMakeFiles/visp_python_bindings_stubs] Error 1
make[2]: *** [CMakeFiles/Makefile2:32317: modules/python/stubs/CMakeFiles/visp_python_bindings_stubs.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:32210: modules/python/CMakeFiles/visp_python_bindings.dir/rule] Error 2
make: *** [Makefile:8070: visp_python_bindings] Error 2

olivier-roussel, 2025-02-25:
From the line: 
```
ImportError: /home/vispci/visp-ws/visp-build-bindings/lib/libvisp_core.so.3.6: undefined symbol: _ZN8vpMatrixaSEd
```
It looks that you have multiple installations from Visp, and here outside of your conda environment in `/home/vispci/visp-ws/`), instead of your current conda environment build dir `/home/vispci/miniforge3/envs/visp-conda-ws`.
So here, for some reasons, installations got mixed up (maybe some environment variables).

Could you remove your other installation in `/home/vispci/visp-ws` ? 
Did you redefine some environment variables in your conda environment `visp-conda-ws` ? And could you provide us the output of your `env` command ?

olivier-roussel, 2025-02-25:
Also, could you share us what did you do to solve your first problem with X11 ?

GavinYang5, 2025-02-25:
Yes, I originally used a dockerfile to build visp, but then I found out that I needed a python bind, so I configured the conda environment and compiled the python wrapper in the docker container.
Regarding X11, I copied the missing files from /usr/include/X11/ to /home/vispci/miniforge3/envs/visp-conda-ws/include/X11/.
The final error was due to some incompatibility with Python 3.12. I downgraded to Python 3.10 and recompiled, and it worked fine.

closed 2025-02-25 by GavinYang5

olivier-roussel, 2025-02-25:
Ok thanks for the feedback.
I'm surprised about the Python version incompatibility as we did not raise problems with Python 3.12 and from your logs he bindings are generated for Python 3.13 (`/home/vispci/miniforge3/envs/visp-conda-ws/lib/python3.13`, python installed: `
python                    3.13.2          hf636f53_101_cp313    conda-forge`).
There was cleary some mixed installations so I still recommend to clear this to avoid any problems in the future.