lagadic/visp #1601: Questions about object 3D model files

GavinYang5, 2025-02-27:
I have noticed that the model-based tracking module supports importing 3D model files with the .cao and .wrl extensions. Regarding the .wrl files, I need to download Coin, but I failed to do so.
I would like to ask, what kind of format is .cao? I only have .obj 3D model files; how should I use them?
I hope to receive your reply. Thank you.

GavinYang5, 2025-02-27:
[image]
I want to know how the red lines of the castle in this video are defined?

fspindle, 2025-02-27:
.cao is a format specific to ViSP which is described [here](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic.html#mb_generic_advanced_cao). In the video you are pointing to, the red lines correspond to the projection of the visible faces defined in the .cao file.

As you mention, .wrl files can be used instead of .cao files, but Coin3D must be installed in order to parse the .wrl model file. On Ubuntu, Coin3D can be easily installed with `sudo apt install libcoin-dev`.

The `visp_mbt` module does not support any other cad model format than .cao or .wrl. In both cases, you need to ensure that the cad model contains only visible external edges (like the red lines in the video). You cannot model a face with a set of triangles.

Note that there is this [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-CAO-editor.html) that explains how to use blender to select a set of lines that can be exported as .cao. Honestly, I haven't tried this tool in a while. I usually open the .obj file in blender, manually identify the coordinates of a 3D point and then manually edit the .cao file step by step to build the model of the object by hand. I know it's a pain.

You should be aware that we have introduced the `visp_rbt` module, which is a major evolution of `visp_mbt`. The RBT (Render Based Tracker) tracker requires Panda3D 3rd party (see installation instructions [here](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-panda3d.html#tutorial-panda3d-install)). It allows to tracks objects in .obj format. For the moment, there is no documentation available in ViSP explaining how to use this new tracker. We only have a few c++ tutorials at https://github.com/lagadic/visp/tree/master/tutorial/tracking/render-based
The plan in to add the documentation soon.

GavinYang5, 2025-02-28:
Yes, manually constructing the .cao file is very painful, and I'm just a beginner with VISP and Blender, so this is very difficult for me. For my scene, I only have high-precision point clouds, and I don't actually have an OBJ model. In this case, do I need to select some points, lines, and surfaces based on the scene's point cloud? The objects I need to track may not just be one object; they could be some lines, surfaces, and points, similar to the castle example. Do you have any other good suggestions? Thank you very much.

fspindle, 2025-02-28:
In this kind of situation, I generally use a ruler to build the .cao model. This is what I did for the castle where you can see that the CAD model is very simple (a planar surface for the bastion, a cube for the dungeon, a sloping surface for the right-hand slide).

If you want to test RBT tracker there is this [tutorial](url) that allows to build the .obj thanks to NeRF. 

I forgot to mention an other algorithm that we have in ViSP that is a wrapper over Megapose and that allows to consider complex .obj files. A tutorial is available [here](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-megapose.html).

GavinYang5, 2025-02-28:
Can the .cao obtained by measuring with a ruler achieve sub-millimeter accuracy? Because I intend to use it for visual servoing on a robot, where the accuracy requirement is sub-millimeter.
I haven't tried RBT yet, but I do have a method to reconstruct point clouds into .obj.
Regarding Megapose, I actually learned about VISP through it. In fact, there are now many pose estimation models that are better than Megapose; however, these models cannot achieve very fast inference times, so I haven't considered them for now.

fspindle, 2025-02-28:
Yes, you can achieve sub-millimeter accuracy because you are closing the loop by visual servo. Even if the CAD model is not sub-millimeter precise, or if you have calibration intrinsic and extrinsic errors. In that case, the best would be to learn the desired features corresponding to your final robot position.

Sure using AI based methods will slow down the visual servo loop. Depends on the application.

fspindle, 2025-03-10:
If the initial pose is not too far from the real pose, the tracker should be able to handle this initial error without affecting later tracking results ans servo.

closed 2025-03-14 by fspindle