lagadic/visp #1602: make error

qian26, 2025-03-04:
Hi, I have met some problems while using visp on Ubuntu22.04.
I encountered some errors while building VISP “make -j$(nproc)”, as shown like this:

[ 18%] Building CXX object modules/imgproc/CMakeFiles/visp_imgproc.dir/src/vpCircleHoughTransform_circles.cpp.o
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp: In member function ‘void vpDisplayX::Impl::clearDisplay(const vpColor&, unsigned int, unsigned int)’:
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp:84:7: error: ‘XSetWindowBackground’ was not declared in this scope
   84 |       XSetWindowBackground(display, window, x_color[color.id]);
      |       ^~~~~~~~~~~~~~~~~~~~
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp:90:7: error: ‘XAllocColor’ was not declared in this scope
   90 |       XAllocColor(display, lut, &xcolor);
      |       ^~~~~~~~~~~
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp:91:7: error: ‘XSetForeground’ was not declared in this scope; did you mean ‘XNForeground’?
   91 |       XSetForeground(display, context, xcolor.pixel);
      |       ^~~~~~~~~~~~~~
      |       XNForeground
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp:94:5: error: ‘XClearWindow’ was not declared in this scope; did you mean ‘XNClientWindow’?
   94 |     XClearWindow(display, window);
      |     ^~~~~~~~~~~~
      |     XNClientWindow
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp:96:5: error: ‘XFreePixmap’ was not declared in this scope; did you mean ‘XYPixmap’?
   96 |     XFreePixmap(display, pixmap);
      |     ^~~~~~~~~~~
      |     XYPixmap
/home/cqy/visp-ws/visp/modules/gui/src/display/vpDisplayX.cpp:98:14: error: ‘XCreatePixmap’ was not declared in this scope
   98 |     pixmap = XCreatePixmap(display, window, width, height, screen_depth);
      |              ^~~~~~~~~~~~~

What should I do to make ViSP successfully?

fspindle, 2025-03-04:
We need more info!
- Which ViSP version ?
- What is the output of
   ```
   sudo apt install locate
   sudo updatedb
   locate libvisp_core
   ```
- did you try a fresh build?
  ```
  cd /home/cqy/visp-ws/
  rm -rf visp-build
  mkdir visp-build
  cd visp-build
  cmake ../visp
  make -j$(nproc)
  ```
- What is the content of `ViSP-third-party.txt` file?

qian26, 2025-03-04:
- Which ViSP version ?
I downloaded it using Git Clone, the latest version 3.6.0

- The output is 

[image]

- I have tried several times for fresh build, and I have got the same results, which is error

[image]

- ViSP-third-party.txt is 

==========================================================
General configuration information for ViSP 3.6.1

  Version control:               v3.6.0-2260-g239fe0852

  Platform:
    Timestamp:                   2025-03-04T13:22:32Z
    Host:                        Linux 5.15.167.4-microsoft-standard-WSL2 x86_64
    CMake:                       3.22.1
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/gmake
    Configuration:               Release

  System information:
    Number of CPU logical cores: 12
    Number of CPU physical cores: 6
    Total physical memory (in MiB): 7873
    OS name:                     Linux
    OS release:                  5.15.167.4-microsoft-standard-WSL2
    OS version:                  #1 SMP Tue Nov 5 00:21:55 UTC 2024
    OS platform:                 x86_64
    CPU name:                    Unknown P6 family
    CPU description:             6 core 12th Gen Intel(R) Core(TM) i5-12490F
    Is the CPU 64-bit?           yes
    Does the CPU have FPU?       yes
    CPU optimization:            SSE2 SSE3 SSSE3
    CPU endianness:              little-endian

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                /usr/bin/c++  (ver 11.4.0)
    C++ flags (Release):         -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C++ flags (Debug):           -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C flags (Debug):             -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    Linker flags (Release):
    Linker flags (Debug):
    Use cxx standard:            17

  ViSP modules:
    To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 java
    Enable visp namespace:       no
    Enable explicit keyword:     no

  Python 3:
    Interpreter:                 /usr/bin/python3.10 (ver 3.10.12)

  Java:                          
    ant:                         no
    Java:                        yes (ver 11.0.26)
    JNI:                         /usr/lib/jvm/default-java/include /usr/lib/jvm/default-java/include/linux /usr/lib/jvm/default-java/include
    Java wrappers:               no

[... 133 lines cut ...]
    Use TensorRT:                no

  Documentation: 
    Use doxygen:                 yes (ver 1.9.1)
    \- Use mathjax:              no

  Tests and samples:
    Use catch2 (built-in):       yes (ver 3.7.1)
    Tests:                       yes
    Apps:                        yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes
    Dataset found:               no

  Library dirs:
    Eigen3 dir:                  /usr/share/eigen3/cmake
    OpenCV dir:                  /usr/lib/x86_64-linux-gnu/cmake/opencv4
    Ogre inc dir:                /usr/include/OGRE/
    Ogre Main lib:               /usr/lib/x86_64-linux-gnu/libOgreMain.so
    Ogre plugin dir:

  Install path:                  /usr/local

==========================================================

fspindle, 2025-03-04:
Your build conflicts with `visp 3.5.0` already installed in `/usr/lib/x86_64-linux-gnu`.
You should remove `visp 3.5.0` installation.
If this version was installed from the `libvisp-dev` ubuntu package, run `sudo apt remove libvisp-dev`. It should be enough.
Then try a fresh visp build from source.

qian26, 2025-03-05:
- I have used `sudo apt remove libvisp-dev`, and the result of `locate libvisp_core` is

[image]

- But when I bulid it using 

`cd /home/cqy/visp-ws/
rm -rf visp-build
mkdir visp-build
cd visp-build
cmake ../visp
make -j$(nproc)`
I have got the same result

[image]
So I dont know what to do

- The `ViSP-third party.txt` is

==========================================================
General configuration information for ViSP 3.6.0

  Version control:               unknown

  Platform:
    Timestamp:                   2025-03-05T03:18:48Z
    Host:                        Linux 5.15.167.4-microsoft-standard-WSL2 x86_64
    CMake:                       3.22.1
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/gmake
    Configuration:               Release

  System information:
    Number of CPU logical cores: 12
    Number of CPU physical cores: 6
    Total physical memory (in MiB): 7873
    OS name:                     Linux
    OS release:                  5.15.167.4-microsoft-standard-WSL2
    OS version:                  #1 SMP Tue Nov 5 00:21:55 UTC 2024
    OS platform:                 x86_64
    CPU name:                    Unknown P6 family
    Is the CPU 64-bit?           yes
    Does the CPU have FPU?       yes
    CPU optimization:            SSE2 SSE3 SSSE3

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                /usr/bin/c++  (ver 11.4.0)
    C++ flags (Release):         -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C++ flags (Debug):           -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C flags (Debug):             -Wall -Wextra -fopenmp -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    Linker flags (Release):
    Linker flags (Debug):
    Use cxx standard:            17

  ViSP modules:
    To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 java

  Python (for build):            /usr/bin/python3.10

  Java:                          
    ant:                         no
    JNI:                         no

  Build options: 
[... 112 lines cut ...]
    Use Simd (built-in):         yes (ver 4.9.109)

  DNN: 
    Use CUDA Toolkit:            no
    Use TensorRT:                no

  Documentation: 
    Use doxygen:                 yes
    \- Use mathjax:              no

  Tests and samples:
    Use catch2 (built-in):       yes (ver 2.13.7)
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes
    Dataset found:               no

  Library dirs:
    Eigen3 include dir:          /usr/share/eigen3/cmake
    OpenCV dir:                  /usr/lib/x86_64-linux-gnu/cmake/opencv4

  Install path:                  /usr/local

==========================================================

fspindle, 2025-03-05:
Before using `locate` with for example `locate libvisp_core` you should always run `sudo updatedb` to update the file list.

It seems that `sudo remove libvisp-dev` doesn't remove completely visp 3.5.0. This installation can be removed manually.

If one of the following folders/files exist remove them
```
ls /usr/include/visp3
ls /usr/share/visp-3.*
ls /usr/bin/visp-*
ls /usr/lib/libvisp_*
ls /usr/lib/cmake/visp
ls /usr/lib/x86_64-linux-gnu/libvisp_*
ls /usr/lib/x86_64-linux-gnu/cmake/visp
```
same in `/usr/local`
```
ls /usr/local/include/visp3
ls /usr/local/share/visp-3*
ls /usr/local/bin/visp-*
ls /usr/local/lib/libvisp_*
ls /usr/local/lib/cmake/visp
```

and try a fresh build.

qian26, 2025-03-05:
I reinstalled a new `Ubuntu-22.04` environment, installed `ros-humble`, `gazebo`, and `CUDA-11.8`, without installing any other redundant packages. Then I built the source code of the visp according to the official instructions, and then reported the same error, but this time I was in a new environment at this time, and did not install any packages that might cause exceptions.

[image]

[image]

[image]

fspindle, 2025-03-05:
Then there's something odd about the detection of Xlib.h.

What is the output of:
```
sudo updatedb
locate X11/Xlib.h
locate X11/Xutil.h
grep XAllocColor /usr/include/X11/Xlib.h
```

qian26, 2025-03-05:
After adopting your suggestion, I successfully built it.  THANK YOU SO MUCH!!!!!!!!!

By the way, I saw on the official website that 

[image]

What impact will it have if I use `cmake ../visp -DUSE_X11=OFF`? Will it prevent me from opening a window to display images?

fspindle, 2025-03-05:
To display images we can use either X11, OpenCV or GTK-2.
You should install opencv and use vpDisplayFactory class that makes abstraction of the 3rd party used for display.

qian26, 2025-03-05:
OK!!!! 
THANK YOU AGAIN FOR SAVING MY LIFE!!!!

closed 2025-03-14 by fspindle