lagadic/visp #1674: Franka binding missing

bakumangit, 2025-05-15:
When I try to get pthon api about franka,I found robot module does not contain anything about Franka while other robots like  Afma4,Afma6 existed.I also found https://github.com/lagadic/visp/pull/1673  and try it,but nothing happened.Could it be fixed?

fspindle, 2025-05-15:
Can you attach your `ViSP-third-party.txt` file located in build folder

fspindle, 2025-05-15:
You don't have Franka bindings because `libfranka` is not detected. This could be seen in `ViSP-third-party.txt` file.

```
Use Franka:                  no
```

Did you install [libfranka](https://github.com/frankaemika/libfranka)? Which version?

bakumangit, 2025-05-15:
I have install libfranka 0.9.2.However, I install it before I use conda to create python3.8 environment,did it lead to this problem?

olivier-roussel, 2025-05-15:
@bakumangit You should most compile & install libfranka from your conda environment. So first create & activate it, then compile libfranka from sources and ensure it will be installed in your conda environment, i.e. something like:
```cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DBUILD_TESTS=OFF ..```
Also, take care to install first cmake and compilers from conda in your conda environment, something like:
```conda install cmake cxx-compiler ninja``` (or replace `ninja` with `make` if you are using it).

olivier-roussel, 2025-05-15:
Also, as you should compile libfranka in your conda environment, please do not install libfranka dependencies from `apt`, but use also conda packages, i.e.:
```conda install pinocchio eigen fmt poco```

olivier-roussel, 2025-05-15:
Sorry I just realized you need libfranka v0.9.2, so you need fewer dependencies to install:
```conda install eigen poco```
It seems that with this libfranka old version, it may work also if you compile it using `apt` installed version of `eigen` and `poco`, as indicated in the doc [here](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-franka-pbvs.html#franka_prereq_libfranka).

bakumangit, 2025-05-15:
> On our side we are able to interact with the Franka robot using the changes I introduced in [#1673](https://github.com/lagadic/visp/pull/1673) and libfranka 0.9.2.
> 
> What is the output of:
> 
> ```
> $ sudo updatedb
> locate FrankaConfig.cmake
> ```
This is the result:
/home/robotman/libfranka/build/FrankaConfig.cmake
/home/robotman/libfranka/build_conda_python3.10/FrankaConfig.cmake
/home/robotman/libfranka/cmake/FrankaConfig.cmake.in
/opt/ros/noetic/lib/x86_64-linux-gnu/cmake/Franka/FrankaConfig.cmake

fspindle, 2025-05-15:
You should try setting `Franka_DIR` env var.
```
$ export Franka_DIR=/home/robotman/libfranka/cmake
$ cmake ../visp
```
and check in `ViSP-third-party.txt` if Franka is detected

bakumangit, 2025-05-15:
> You should try setting `Franka_DIR` env var.
> 
> ```
> $ export Franka_DIR=/home/robotman/libfranka/cmake
> $ cmake ../visp
> ```
> 
> and check in `ViSP-third-party.txt` if Franka is detected

I try to reinstall libfranka and it worked.I install it in the default path /usr/local and then Franka is detected.It is so kind of you to help me solve this problem.All of your work is great.

fspindle, 2025-05-16:
PR #1673 is now fully merged in master branch. There is an short example in https://github.com/lagadic/visp/blob/master/modules/python/examples/franka-connexion.py

closed 2025-05-16 by fspindle